echostack-org/echopype #426: xarray requirements

mbdunn, 2021-08-17:
When installing echopype the xarray package that is automatically downloaded is v.0.16 but needs to be v0.17+ to avoid errors with combine-attrs.

emiliom, 2021-08-17:
@mbdunn we're aware of this issue and have implemented a fix. We're wrapping up a new release (0.5.3) that should be out within a few days (@leewujung can confirm this!), and in that release the xarray version installed should be the latest stable version -- currently 0.19.0. Stay tuned!

leewujung, 2021-08-18:
Yes! Once we get 0.5.3 released -- should be very soon, we are just waiting for a small docstring fix -- this should be good. If you want to get that now, feel free to pull the `dev` branch since that particular PR (which unpinned versions of all dependencies) is already merged and functional.

mbdunn, 2021-08-19:
Awesome! Great that it is already taken care of. I'll close the issue :)

closed 2021-08-19 by mbdunn