echostack-org/echopype #442: Migrate master branch to main

emiliom, 2021-09-23:
*I'm taking advantage of an email from IOOS with helpful background and instructions to create an issue as a reminder to ourselves to make this transition in the not too distant future.*

The Software Freedom Conservancy released a statement (https://sfconservancy.org/news/2020/jun/23/gitbranchname/) last year detailing why they are encouraging projects to shift from "master" to "main" and lay out some good reasons for it; GitHub has gone the same direction (https://github.blog/changelog/2020-10-01-the-default-branch-for-newly-created-repositories-is-now-main/), as have several other git hosters.

Once a repository has been changed to use `main` as the default branch, if you have a local clone, you can update it by running:

```bash
git branch -m master main
git fetch origin
git branch -u origin/main main
git remote set-head origin -a
```

leewujung, 2021-09-23:
Ah this is great! Thank you @emiliom ! We can probably make this change when we do the next release, or the next major release?

leewujung, 2021-10-02:
Done!

closed 2021-10-02 by leewujung