echostack-org/echopype #452: Change both CI actions flags to be based on PR titles

emiliom, 2021-10-04:
[Currently](https://echopype.readthedocs.io/en/stable/contributing.html#continuous-integration-github-actions), the CI can be skipped via the commit string "[skip ci]", and the PR label `Needs Complete Testing` is used to force the running of all CI tests.

For consistent usage and convenience, change this so that both are implemented as PR title strings: "[skip ci]" and "[all tests ci]".

From @lsetiawan: Looks like there's a "title" value that I can grab and check, https://docs.github.com/en/rest/reference/pulls

emiliom, 2021-11-07:
Once this is implemented (maybe soon after 0.5.5 release?), let's remember to update the docs, https://echopype.readthedocs.io/en/stable/contributing.html#continuous-integration-github-actions

emiliom, 2022-06-04:
@lsetiawan I believe this isn't implemented yet, right? Would it be easy to add in the near term?

lsetiawan, 2022-06-06:
It's not implemented yet. Sorry I missed this issue previously. I think that can be added soon. I'll take a look. Thanks for putting it in my radar again.

emiliom, 2022-06-06:
> Sorry I missed this issue previously. 

No worries! We have a lot to keep track of and prioritize.

> I think that can be added soon. I'll take a look. Thanks for putting it in my radar again.

Awesome. Thanks!

emiliom, 2022-06-10:
This change has been implemented (PR #721). But let's leave this issue open for now as a reminder to [update the documentation](https://echopype.readthedocs.io/en/stable/contributing.html#continuous-integration-github-actions)

emiliom, 2022-07-08:
We're now tracking the update to the documentation in #757. Closing

PR #721 (2022-06-09): Update pr action to use title [all tests ci]
This PR moves triggers for running all tests to look for `[all tests ci]` flag or `[skip ci]` flag rather than just label. This resolves #452.
closed 2022-07-08 by emiliom