echostack-org/echopype #469: Zenodo badge leads to a very old version (0.4.1).

lsetiawan, 2021-11-01:
The current zenodo DOI leads to a very old version of 0.4.1. Rather than the latest 0.5.4. This needs to be changed. Echopype does have a `concept doi` that leads to the latest version of the software everytime. Maybe we can use that instead, or create a way to update the zenodo badge at each release.

**Zenodo Concept DOI**

[image]

leewujung, 2021-11-01:
Oh I didn't realize that it's not linked to the general DOI. Yes I think using that would work. I'll do a PR. Thanks for noticing it! 😄

leewujung, 2021-11-04:
This is again addressed by #473 .

leewujung, 2021-12-03:
Let's close this now to reduce clutter. :)

PR #472 (2021-11-02): update zenodo link and badge to use general doi
This resolves #469
closed 2021-12-03 by leewujung