echostack-org/echopype #478: Improve docstring consistency

leewujung, 2021-11-05:
Right now in the docstring there are both double backquotes ("````")  and single backquotes ("``")  used to indicate inline code section. They are rendered differently in the RTD docs:
- double backquoted text is rendered as inline code
- single backquoted text is rendered as italicized text

Perhaps we have some RTD config issue? Or we need to pick one and stick with it?

emiliom, 2021-11-29:
Function parameters that are optional are not indicated as such in the docstrings in a consistent way across the board. One example where the string "optional" is used is in [dev/echopype/calibrate/api.py](https://github.com/OSOceanAcoustics/echopype/blob/dev/echopype/calibrate/api.py) (once PR #477 is merged). Using a consistent scheme across echopype would be helpful.

leewujung, 2022-10-12:
Close because this is duplicate of  #593 -- an indication that this has dragged too long!

closed 2022-10-12 by leewujung