echostack-org/echopype #484: Consolidate definition and use of `DEFAULT_CHUNK_SIZE`

leewujung, 2021-11-09:
`DEFAULT_CHUNK_SIZE` is curiously declared in two different places: `echopype/convert/set_groups_base.py` and `echopype/convert/api.py`. 

It is only used in 1 place for single file handling (`SetGroupsEK60.set_platform`) and does not seem to be used in any other places (!)

Strangely (but nicely), it is used in `convert_combine.perform_combination` which is used by `ep.combine_echodata`.

These should be resolved when we move to more performance-related work.

b-reyes, 2022-10-12:
@leewujung I did a quick search and we are not really using `DEFAULT_CHUNK_SIZE` anymore. It is used in a couple of spots, but they are currently commented out. Also, it is still defined in two places. I do not plan on using this in `combine_echodata`.

leewujung, 2022-10-12:
So we can just remove this completely then?

emiliom, 2022-10-12:
Looks like we should

b-reyes, 2022-10-12:
> So we can just remove this completely then?

If you are asking if we can remove `DEFAULT_CHUNK_SIZE`, then yes. We can definitely remove this.

leewujung, 2022-10-12:
Yes. Could you please submit a PR to remove that? Thanks!

b-reyes, 2022-10-13:
In PR #848 we have removed `DEFAULT_CHUNK_SIZE`, so this issue is no longer necessary. I will go ahead and close it.

closed 2022-10-13 by b-reyes