echostack-org/echopype #506: Remove old API

leewujung, 2021-12-10:
Currently we issue deprecation warnings for the old API. Let's remove it completely in the next major release.

leewujung, 2022-03-25:
(Note for myself)
Also remove obvious unused attributes:
https://github.com/OSOceanAcoustics/echopype/blob/a18ec08c543292693e4bbff1ae02cce9e3b3a5d5/echopype/convert/parse_base.py#L43

leewujung, 2022-03-25:
Remove unused comments:
https://github.com/OSOceanAcoustics/echopype/blob/a18ec08c543292693e4bbff1ae02cce9e3b3a5d5/echopype/convert/set_groups_ek80.py#L601-L605

emiliom, 2022-03-28:
I suggest we prioritize this issue before convention-related changes in 0.6.0. Removing the old API's will make the convention-related changes a bit less burdensome, because there will be less code to go through. We've already run into the drag of having to modify the code involving the deprecated API's.

leewujung, 2022-03-28:
Sounds good. Let me get on it as soon as my environment is back... 😅

closed 2022-03-29 by leewujung