echostack-org/echopype #519: Move Beam group to be a Sonar subgroup

emiliom, 2021-12-15:
Move `Beam` group from the current top level to one (or more?) `Sonar` subgroups, to adhere to the convention. Look up our discussion of Beam groups in the echopype paper to remember whether we see a possible need to ever have more than one Beam group.

**TODO:** Add a link/reference to the SONAR-netCDF4 convention section that discusses this

leewujung, 2021-12-16:
Note: also consider #490 which is related to having >1 transducer/channel of the same frequency in the echosounder setup.

emiliom, 2022-01-27:
`Sonar` beam sub-groups are discussed in the [SONAR-netCDF4 v1.0 report](https://www.ices.dk/sites/pub/Publication%20Reports/Cooperative%20Research%20Report%20(CRR)/CRR341.pdf) in section 2.10.6, page 14:

> Data from each beam mode (e.g. horizontal and vertical beam modes) are stored in subgroups under the `/Sonar` group (see Table 8). The form of the backscatter data can vary between different sonar systems. For example, some provide a complex-valued amplitude, while others provide a real- or integer-valued amplitude. Variable definitions for data from split-aperture systems are not currently specified.

> Include as many subgroups as necessary for different beam groups. Use unique group names, preferably of the form `Beam_groupX` where X is an integer.

A critical, distinguishing global attribute for each Beam group is `beam_mode`:

> Mode of the beam in this sub-group, taken from the defined vocabulary of: “vertical” (a set of beams that form a vertical slice through the water), ”horizontal” (a set of beams that form a nominally horizontal plane through the water), and ”inspection” (a set of beams with arbitrary pointing directions)

A Beam group may have more than one `beam`, discriminated through the `beam` dimension.

emiliom, 2022-01-27:
### SONAR-netCDF4 Version 2 draft

The rendered (updated daily, I think) version of the SONAR-netCDF4 Version 2 draft is available [here](https://htmlpreview.github.io/?https://github.com/ices-publications/SONAR-netCDF4/blob/master/Formatted_docs/crr341.html). Version 2 includes some changes to the `Beam` group. They are partly summarized in [7.1. Significant changes from version 1 to version 2](https://htmlpreview.github.io/?https://github.com/ices-publications/SONAR-netCDF4/blob/master/Formatted_docs/crr341.html#_significant_changes_from_version_1_to_version_2).

The `Sonar` group hierarchy shown below is constructed from what's in [3.2. Hierarchical structure](https://htmlpreview.github.io/?https://github.com/ices-publications/SONAR-netCDF4/blob/master/Formatted_docs/crr341.html#_hierarchical_structure) and in [3.10.6. Proposed ADCP additions](https://htmlpreview.github.io/?https://github.com/ices-publications/SONAR-netCDF4/blob/master/Formatted_docs/crr341.html#_proposed_adcp_additions):

[`Sonar`](https://htmlpreview.github.io/?https://github.com/ices-publications/SONAR-netCDF4/blob/master/Formatted_docs/crr341.html#_sonar_group) – contains the main data models in the convention; groups under Sonar are used for storing the data output from the sonars, as well as processed sonar data and interpretation masks;
- `Beam_groupX` – contains ping based backscatter; the group supports multiple sensors of the same type, e.g. multiple frequencies;
  - `ADCP` - This group contains the ADCP beam data, single ping calculations and associated metadata;
    - `Mean_current` - This group contains the calculated averaged current velocity data and associated metadata;
- `GriddedX` – contains gridded backscatter data; the grid is similar across multiple frequencies and can be used for fine scale grids as well as coarser resolution typically as a basis for the reports;
- `InterpretationX` – contains the interpretation regions; the intepretation mask is independent from the backscatter groups;

emiliom, 2022-01-30:
In #545 I've implemented a draft version of the `Beam` group move to the `Sonar/Beam` subgroup . 

But there's a larger question we'll need to address (see above,https://github.com/OSOceanAcoustics/echopype/issues/519#issuecomment-1023573467), of whether our interpretation of SONAR-netCDF4 v1.0 of Beam groups as mappable to the frequency coordinate in echopype is correct. Plus double checking how changes proposed in v2.0 impact our use of the Beam group

emiliom, 2022-01-31:
Just realized that we should also address the `Beam_power` group. It's used only with EK80. But I don't know what it is or how it maps to the v1.0 convention. Will need to discuss.

leewujung, 2022-01-31:
@emiliom : the `Beam_power` group was my ad-hoc (!) creation to accommodate EK80 datasets that contain both complex time series samples (from BB or CW mode) and power/angle samples (from CW mode). This is because the number of samples are drastically different (complex samples are much large as they are sampled at a much higher freq), and the complex samples also have an extra dimension (currently that is called `quadrant`, which can be viewed as just as a separate transducer, or a "beam").

Thinking about the `Beam_groupX` setup, it seems that it would work to just have the power/angle samples stored as one o the beam groups, in parallel with the complex samples. We would need to figure out how to index these.

Transmit signal type: BB - broadband; CW - continuous wave (narrowband)

emiliom, 2022-02-09:
> the `Beam_power` group was my ad-hoc (!) creation to accommodate EK80 datasets that contain both complex time series samples (from BB or CW mode) and power/angle samples (from CW mode). This is [...]

The second part, the  extra dimension (`quadrant`, which as we've discussed will be renamed to `beam`) is not an issue per se, since we'll be adding a `beam` dimension to the `Beam` group that hasn't had it b/c it was implicit. Right? And the issue of number of samples being drastically different bears resemblance to similar cases we've discussed; in that case the core challenge is that our regular gridded structure makes this sparse situation *very* inefficient, right? But is there ultimately a conceptual, clear reason, based on the SONAR-netCDF4 definition of a `Beam` group, why the power/angle samples and the complex samples should go in separate groups?

> Thinking about the `Beam_groupX` setup, it seems that it would work to just have the power/angle samples stored as one o the beam groups, in parallel with the complex samples. We would need to figure out how to index these.

What do you mean by "how to index these"?

leewujung, 2022-02-09:
> is there ultimately a conceptual, clear reason, based on the SONAR-netCDF4 definition of a Beam group, why the power/angle samples and the complex samples should go in separate groups?

By "separate groups", do you mean to put data collected by the same frequency channel but under different settings (CW with power/angle samples, CW in complex samples, BB in complex samples) in the same group?  I think that will cause memory problem unless we use sparse representation to circumvent the NaN padding. Also, as of now, Power samples live in backscatter_r only and complex samples live in both backscatter_r and backscatter_i. If we put all of them in the same group and only interleaved by ping_time, it would be hard to slice. 

>> Thinking about the Beam_groupX setup, it seems that it would work to just have the power/angle samples stored as one o the beam groups, in parallel with the complex samples. We would need to figure out how to index these.

> What do you mean by "how to index these"?

I was referring to how to tell which `Beam_groupX` contains what type of data. For example, what determines the ordering 1, 2, 3, ... ?  And if we use different `Beam_groupX` to store separately complex and power/angle data that were collected by the same channel at different time with different settings, the 1, 2, 3, ... does not give use a simple way to figure out which group contains what. It seems counter productive if we need to load the group before can tell whether it is what we need.

emiliom, 2022-03-04:
Just a note to emphasize that #567 is also central to this issue.

leewujung, 2022-03-10:
@emiliom : I don't fully understand some of your comments above, but on this part:
>  if in practice there'll ever be a case with 3 beam groups

I think it depends on what Simrad allows or not in the future. This came up in our conversation with @gavinmacaulay also, about whether changing the type of the samples being collected forced the system to stop the current file and start a new one.

We know based on the test files we have that it is possible to have 1 .raw file containing both the CW-power/angle data and BB-complex data. These go into the `ed.beam_power` and `ed.beam` currently, which is the hard-wiring we want to change. If we go to using the `ed["Sonar/Beam_groupX"]` pattern and deprecate the current `ed.sonar` access pattern then the number won't be limited.

About the access pattern: my understanding from our last group discussion was that we would use open_datatree to open the file, which then allows using something like `ed["Sonar"]` (with or without `.ds`) to access data. Though looking at this again now I don't remember what we said about how to handle this coexisting with `ed.sonar`. with deprecate the pattern `ed.sonar` altogether in a minor release? @lsetiawan @b-reyes @imranmaj help!

imranmaj, 2022-03-10:
We can change `ed.sonar` to call `ed["Sonar"]` under the hood before we deprecate it

emiliom, 2022-03-10:
Thanks @leewujung and @imranmaj! Sorry if my comments weren't very clear. I should clarify that those comments are not focused on the `ed.<group>` vs `ed["<group>"]` mappings. For that, I think the discussion should take place in #567. My comments, and PR #574, are focused on what @leewujung described as step 1 in her comment above (a minute ago). Of course, these things are all interrelated.

So, for my focus the hard-wired beam group limit I'm referring to (and that I've partially addressed in #574) concerns especially this:

> An immediate challenge will be that this hard-wiring currently starts in the explicit listing of groups in [1.0.yml](https://github.com/OSOceanAcoustics/echopype/blob/dev/echopype/echodata/convention/1.0.yml) and its usage in [EchoData.__setup_groups()](https://github.com/OSOceanAcoustics/echopype/blob/dev/echopype/echodata/echodata.py#L122). We'll need to refactor this pattern so that the number of groups is not universally fixed in a single place.

leewujung, 2022-03-10:
Thanks for the clarification @emiliom ! Sorry I was confused! I will now work on your #574 .

emiliom, 2022-04-09:
I think we'll be able to close this after PR #611 is merged

emiliom, 2022-04-12:
#611 has been merged, so we can close this issue. Woo-hoo!

closed 2022-04-12 by emiliom