echostack-org/echopype #522: roll is both method and DataArray

rolandproud, 2021-12-17:
Hi, in the latest pip version, roll is both an xarray method and DataArray (e.g. EchoData.platform.roll). This may cause issues - there are easy work arounds but it may be worth changing this.

Cheers,

Roland

leewujung, 2021-12-18:
@rolandproud : Hmm, the roll is a variable defined in the convention, so we probably don't want to change that. What are the use cases you see can cause issues? Like is someone wants to do `EchoData.platform.roll.roll`?

emiliom, 2021-12-21:
The existence of `roll` to indicate two different things is unfortunate and indeed potentially confusing; but as @leewujung said, it's not something under our control, since those meanings are external (xarray package and SONAR-netCDF4 convention) and we shouldn't mess with them. At best we can add some notes to the documentation about this potential confusion.

rolandproud, 2021-12-23:
OK fair enough. Just thought I would point this out in case you could think of a quick and easy fix!

closed 2021-12-23 by rolandproud

leewujung, 2021-12-23:
Sounds good! One thing I thought of when seeing this is that there's `range` in the Sv output and the python reserved word `range`. The one in the Sv output probably can be changed to `sonar_range` to make sure it is unique and understandable. 

Thoughts?

rolandproud, 2022-01-06:
Yes I agree - I would not name anything using the built-in function names. (Note that filter and complex are also built-in functions). For functions from other modules, I would suggest using module.function_name.

leewujung, 2022-01-06:
> For functions from other modules, I would suggest using module.function_name.

This is in general a good practice, but for OO type of methods, that wouldn't work, like in the previous example, `platform.roll` is an xr.DataArray for which `roll` is a method.

My comment above regarding whether to call something `sonar_range` is more along the line of data convention. ;)

rolandproud, 2022-01-06:
Yep - sorry that was just a general comment not in reference to platform.roll.

leewujung, 2022-01-06:
Oh I see. Make sense!