echostack-org/echopype #526: Replace matplotlib with matplotlib-base in requirements

emiliom, 2022-01-12:
`matplotlib` is listed in the [requirements.txt](https://github.com/OSOceanAcoustics/echopype/blob/main/requirements.txt#L2). I suggest we replace it with the lighter weight [matplotlib-base](https://stackoverflow.com/questions/61325817/differences-between-matplotlib-and-matplotlib-base) (matplotlib minus `pyqt`)

Any objections?

leewujung, 2022-01-12:
Sounds good!

emiliom, 2022-01-14:
After realizing that `matplotlib-base` is only available as a conda package (not as a PyPI package), running into build errors in PR #527, and discussing this further, we've decided to go in a different direction: to make `matplotlib` an optional dependency. We'll add matplotlib to https://github.com/OSOceanAcoustics/echopype/blob/main/setup.py#L16, like this:
```python
EXTRA_REQUIRES = {"dev": dev_reqs, "plot": ["cmocean", "matplotlib"]}
```

emiliom, 2022-01-21:
Addressed in #529

closed 2022-01-21 by emiliom