echostack-org/echopype #528: Adding unstable/dev branch warning to readthedocs

imranmaj, 2022-01-13:
Since the "latest" version on readthedocs will now point to the dev branch, it may be a good idea to add a warning to the documentation saying that this version is not stable/may be subject to change.

The warning could be in a [readthedocs warning box](https://sublime-and-sphinx-guide.readthedocs.io/en/latest/notes_warnings.html#warnings). Although the warning should be at least on the home page of the documentation, it would be nice to have it automatically appear on every page (without manually adding it to every page).

leewujung, 2022-07-28:
This one seems outdated now?

emiliom, 2022-07-28:
We haven't resolved this, but I think it's very low priority. The RTD setting referred to in the previous comment **is** actually on (eg, https://echopype.readthedocs.io/en/v0.6.0/). But it doesn't insert the warning on the `latest` version, which was the goal set out in this issue.

So, we're not aware of an existing solution to exactly address this issue; but the issue itself is not a priority, IMHO.

We can close it or leave it open for a while longer, as a reminder. I'll remove it from Milestone 0.6.2

closed 2023-07-25 by leewujung