echostack-org/echopype #534: 💡 Idea: Potentially update echodata object repr to using datatree

lsetiawan, 2022-01-21:
## Overview

Currently, we have a built in echodata object repr that we maintain ourselves that was copied over from the arviz library (https://arviz-devs.github.io/arviz/). Tom Nicholas is developing https://github.com/TomNicholas/datatree a package for a DataTree type of representation. Which looks like 

[image]

this may be a better solution to the echodata object repr in terms of maintainability. However, we might lose some of the nice text that explains what each group is from the convention. Right now the datatree library is not quite ready for use, but good to keep watch 😄

leewujung, 2022-01-21:
Awesome @lsetiawan! Also I didn't know emojis show up in title also! :D

leewujung, 2022-04-22:
🚀 this is now implemented and merged!

closed 2022-04-22 by leewujung