echostack-org/echopype #535: Dependency / requirements reorganization with setup.cfg

emiliom, 2022-01-21:
Recent discussions about optional dependencies (#526, #529) and sphinx dependencies (#529; what should go in `requirements-dev.txt` vs `docs/requirements.txt` vs `.readthedocs.yml`) motivated a discussion of a larger re-think. From @lsetiawan :

> In terms of dependencies in the future, it's better to have all of those docs requirements in the `setup.cfg` .... I will provide more overhaul in listing out dependencies in a separate PR.

Let's discuss the benefits `setup.cfg` brings, and its downsides, if any. One concern I bring is ensuring that local conda dv environments can still be created easily w/o overly complicated steps and instructions. Here are [our current instructions for creating a local dev environment](https://echopype.readthedocs.io/en/stable/contributing.html#installation-for-echopype-development).

My original comments in #529:

> What's the reason for having matplotlib in `.readthedocs.yml` rather than in `docs/requirements.txt`? I ask b/c the latter is easier to incorporate into a local conda env creation statement (for local development & testing). 

> And actually, this leads me to a broader realization: the sphinx plugins and other sphinx-related dependencies are currently listed in **both** [docs/requirements.txt](https://github.com/OSOceanAcoustics/echopype/blob/main/docs/requirements.txt) and [requirements-dev.txt](https://github.com/OSOceanAcoustics/echopype/blob/main/requirements-dev.txt). Wouldn't it make more sense to remove them from `requirements-dev.txt`? Then, if you want to do local sphinx development & testing, you'll have to include `docs/requirements.txt` in the conda env.

leewujung, 2022-07-28:
This one seems outdated now? @lsetiawan @emiliom

emiliom, 2022-07-28:
I agree. There may be some threads here that we'll want to revisit in the future, but keeping this issue open doesn't really help.

leewujung, 2022-07-29:
I'll close this.

closed 2022-07-29 by leewujung