echostack-org/echopype #541: Track down style differences in RTD stable & latest vs tagged versions

emiliom, 2022-01-22:
The style used in the ReadTheDocs (RTD) docs differs between `stable` (the default) and `latest` vs tagged versions :exploding_head:. `stable` and `latest` use a style that, I think, has at least a couple of shortcomings from what I've seen: 
- Lists are not styled as such (with bullets and narrower line separations). For example, see the main, front page
- In API pages, parameter names and type are combined into a single string (wrong, for sure), and the font style for the "Parameters" and "Returns" heading is the same as for the parameters, which seems a bit confusing or misleading. Here is an [example from stable](https://echopype.readthedocs.io/en/stable/api/echopype.utils.uwa.calc_sound_speed.html#echopype.utils.uwa.calc_sound_speed) vs the [same page from the v0.5.5 tagged version](https://echopype.readthedocs.io/en/v0.5.5/api/echopype.utils.uwa.calc_sound_speed.html)

This can't be something we're doing explicitly, unless there's something we can set in `.readthedocs.yml`. At any rate, it'll require some investigation.

emiliom, 2022-01-28:
From @leewujung in https://github.com/OSOceanAcoustics/echopype/pull/530#issuecomment-1024176069:

> @emiliom : this came up after you logged off, @b-reyes thought that he can help with this based on his experience with sphinx.

Woo-hoo! Go for it @b-reyes ! One comment: this smells like it's more related to ReadTheDocs configuration and defaults than to Sphinx proper. But your Sphinx background will be very helpful there.

leewujung, 2022-04-22:
Since we've moved to use jupyter book for docs, it seems that this issue can be closed?

emiliom, 2022-04-22:
I agree, let's close it. If we again notice differences between stable and latest, let's just open a new issue.

closed 2022-04-22 by emiliom