echostack-org/echopype #542: Port echodata/convention/attrs.py to a reusable yaml file

emiliom, 2022-01-27:
I recently created a centralized module that stores attribute values (and related info) from SONAR-netCDF4 v1.0: [echodata/convention/attrs.py](https://github.com/OSOceanAcoustics/echopype/blob/dev/echopype/echodata/convention/attrs.py). We've decided that it should be in a yaml file rather than a Python module, so it can be reused more easily across the board, specially by [echopydantic](https://github.com/OSOceanAcoustics/echopydantic)

lsetiawan, 2022-01-27:
This suggestion came from https://github.com/OSOceanAcoustics/echopype/pull/493#issuecomment-974294527.

emiliom, 2022-02-09:
@lsetiawan I've started working on this. I decided to reuse [convention/1.0.yml](https://github.com/OSOceanAcoustics/echopype/blob/main/echopype/echodata/convention/1.0.yml) and [convention/\_\_init__.py](https://github.com/OSOceanAcoustics/echopype/blob/main/echopype/echodata/convention/__init__.py) by porting [convention/attrs.py](https://github.com/OSOceanAcoustics/echopype/blob/main/echopype/echodata/convention/attrs.py) into `1.0.yml`. That won't impact anything else currently in use, as far as I can tell. However, it means that the yml file would have to be read (via `_get_convention`) every time some portion of it is needed. That seems inefficient. Is it possible (and a good practice) to find a way to "pre load it" always just once, automatically? Or do you think having to read the yml several times is not a big deal?

lsetiawan, 2022-02-10:
> "pre load it" always just once, automatically? Or do you think having to read the yml several times is not a big deal?

I think pre loading it is best, since it seems to me that these metadata will be used in several places. I also agree that your `attrs.py` content can go to the 1.0.yml. This is a great place for that. Thanks for working on this @emiliom.

emiliom, 2022-03-01:
Even though #565 has addressed the core thrust of this issue, I'm inclined to leave it open until we're wrapping up 0.6.0. I think there are related issues that may come up that we can discuss here.

closed 2023-05-24 by emiliom