echostack-org/echopype #568: Autocomplete for EchoData object groups

b-reyes, 2022-02-25:
Currently the groups within the EchoData object are set at runtime, see for example:

https://github.com/OSOceanAcoustics/echopype/blob/17a86e4c87f28d1457c582062e3ac59367b37348/echopype/echodata/echodata.py#L121-L123

However, this does not allow for autocompletion of the groups in IDEs such as PyCharm. The user must run the initialization of the EchoData object first and then the groups will become available. To improve the user experience, I think it would be better if we changed the EchoData object such that autocompletion is enabled without running the initialization. Then for the given EchoData object "ed", a user could do "ed." and see the possible groups of the object, such as top, environment, etc. 

One possible solution to this is to initialize these core groups in __init__(), for example: 

self.top = None
self.environment = None

leewujung, 2022-02-25:
One thought I had was if we can leverage the framework that would be created in echopydantic to put in that template in init. 

On the other hand, if we are just doing `top`, `environment`, `provenance`, `platform`, `sonar`, `vendor`, `annotation` (I don't think we currently have this one?), hard-code them in to enable the type hint in IDE, but at runtime dynamically generating all the associated info and link with the parsed/converted data, seems a workable solution.

leewujung, 2022-02-25:
Also, I think we can take this on in a minor release v0.6.X so that we get all the breaking changes taken care of in v0.6.0 first.

emiliom, 2022-04-09:
I suggest we close this issue, since we're moving away from the EchoData group object referencing

leewujung, 2022-04-10:
> we're moving away from the EchoData group object referencing

This is in principle true in terms of our intention, but we'll still keep a pattern for such referencing in at least v0.6.0.

But, I agree that we can close this issue, and focus on putting together a good documentation page for accessing the different groups in the converted standardized data.

closed 2022-04-25 by emiliom