echostack-org/echopype #571: Populate or fix sonar_software_name and sonar_model for EK60 and maybe other instruments

emiliom, 2022-02-28:
(Originally discussed in https://github.com/OSOceanAcoustics/echopype/issues/210#issuecomment-822642399)

For EK60:

- Populate `Sonar/sonar_software_name` attribute. Right now it's left blank; see [convert/set_groups_ek60.py#L170](https://github.com/OSOceanAcoustics/echopype/blob/dev/echopype/convert/set_groups_ek60.py#L170)
- `Sonar/sonar_model attribute` entry is "ER60". It should be EK60. Currently EK60 `Sonar/sonar_model` is set in [convert/set_groups_ek60.py#L168](https://github.com/OSOceanAcoustics/echopype/blob/dev/echopype/convert/set_groups_ek60.py#L168) as `self.parser_obj.config_datagram['sounder_name']`, so that looks fine, unless the datagram `sounder_name` is parsed incorrectly.

We haven't examined these closely (AFAIK) for EK80, AZFP or other instruments.

These attributes could be examined in parallel with `Sonar/sonar_serial_number`, #212

emiliom, 2022-03-25:
I don't think I'm the right person for this issue, since I'm not well versed with the raw datagrams.

leewujung, 2022-03-25:
I assigned both you and me, because I need to consult you on the naming choice of attributes that we will need to create to take care of the things that may not be in the convention.

emiliom, 2022-03-25:
Ah! Thanks for that clarification. I didn't see that.

emiliom, 2022-05-19:
ER60 is in fact the name of the Simrad software. eg, see https://support.echoview.com/WebHelp/Reference/File_formats/Simrad_data_files.htm

leewujung, 2022-05-26:
We decided to review the detailed content of variables _after_ v0.6.0 now that we have the disruptive type of changes in for v0.6.0.

emiliom, 2022-12-06:
`sonar_model` was already addressed for v0.6.0. See https://github.com/OSOceanAcoustics/echopype/issues/681#issuecomment-1339962902.

In this issue, the only remaining topic is `sonar_software_name`. It may be more convenient to address that question in #681 and close this issue.

closed 2022-12-07 by leewujung