echostack-org/echopype #576: MVBS variable name is Sv

lgarzio, 2022-03-10:
I think the MVBS variable name is incorrectly labeled Sv after doing the following (using echopype version 0.5.6):

ed = ep.open_raw(raw_file, sonar_model='EK60')
ds_Sv = ep.calibrate.compute_Sv(ed)
ds_MVBS = ep.preprocess.compute_MVBS(ds_Sv, range_meter_bin=5, ping_time_bin='20S')

When I open ds_MVBS in xarray I get the following. I think the data variable is probably just mis-labeled as Sv instead of MVBS.

<xarray.Dataset>
Dimensions:    (frequency: 3, ping_time: 113, range: 120)
Coordinates:
  * ping_time  (ping_time) datetime64[ns] 2018-01-16T03:08:20 ... 2018-01-16T...
  * frequency  (frequency) float64 3.8e+04 1.2e+05 2e+05
  * range      (range) float64 0.0 20.0 40.0 60.0 ... 2.34e+03 2.36e+03 2.38e+03
Data variables:
    Sv         (frequency, ping_time, range) float64 ...
Attributes:
    binning_mode:          physical units
    range_meter_interval:  20m
    ping_time_interval:    20S

leewujung, 2022-03-10:
Thanks @lgarzio -- MVBS is _mean_ volume backscattering strength, so it is indeed still Sv. 
Perhaps there should be some explanations somewhere to elaborate this? Do you have suggestions for this??
The pre-processing functions are not really in the documentation at the moment (limited bandwidth...) but we do want to improve it.

I can see that this is also a problem with not having provenance in the output data after processing steps, which we should also improve.

lgarzio, 2022-03-10:
@leewujung I was very confused with my processed files because I did all of the processing a while back, then when I went to plot today I couldn't figure out why Sv was in my MVBS files, hahaha. But yes this makes sense, and I think just adding a comment to the attributes that clarify this is mean volume backscattering strength would definitely suffice. Thanks!

emiliom, 2022-03-10:
Thanks @lgarzio . I agree this is ambiguous and potentially confusing. As @leewujung said, MVBS is `Sv`. But the MVBS dataset that's generated doesn't actually include any attributes that clearly say it's MVBS. The provenance info @leewujung mentions will definitely address this more thoroughly, but yeah, a very useful improvement right now that would be easy to implement is to add a `long_name` attribute to the `Sv` variable, with this value: "Mean Volume Backscattering Strength (MVBS)". Right now the `Sv` variable is missing a `long_name` attribute, which is a bad practice anyways.

leewujung, 2022-03-10:
Related to #221

lgarzio, 2022-03-10:
@emiliom I like that idea of adding the long_name to both this and Sv. The more info you can put in the attrs for users, the better.

emiliom, 2022-03-10:
@leewujung I'm looking into this. Do you know if there's a deliberate reason why the MVBS DataArray attributes are promoted (duplicated) into the Dataset, [here](https://github.com/OSOceanAcoustics/echopype/blob/dev/echopype/preprocess/api.py#L77)? (You can also see the result [here](https://osoceanacoustics.github.io/echopype-examples/echopype_tour.html#compute-mean-volume-backscattering-strength-text-mvbs-and-plot-interactive-echograms-using-hvplot)). I don't see whey that duplication is needed. I'm adding `long_name` to `MVBS.attrs`, but `promote_attrs=True` would lead to the attribute also being a global attribute, which is out of place.

emiliom, 2022-03-10:
While we're focusing on MVBS, I just remembered (or revisited) #481(`ping_time` and `range` are missing user-facing variable attributes). This is a good opportunity to try to address those shortcomings, too.

emiliom, 2022-04-09:
I think we can close this issue. See #594

leewujung, 2022-04-09:
Agreed! I'll go ahead and close this. :)

closed 2022-04-09 by leewujung