echostack-org/echopype #578: Add mechanism to convert `echo_range` to `depth` in calibrated datasets

leewujung, 2022-03-11:
This is derived from #259.

We decided that it is more flexible to have the `echo_range` and `range_offset` to be separate in the calibrated dataset (the Sv). This is addressed in #577.

However, it is necessary to have a mechanism for users to convert and combine these values into "depth." For example:
- for data from shipboard echosounders, `depth = echo_range + range_offset`, where `range_offset` can the depth of the transducer, or depth of the transducer plus some starting lag when the instrument started collecting data.
- for data from upward looking echosounders, `depth = (sonar_range + range_offset) * cos(theta)`, where `theta` is some tilt angle of the instrument that may or may not be 0.

In these examples there's also the upside down flip depending on whether it is in a downward- or upward-looking configuration.

So, we need to figure out a mechanism to make it easy for users to get things into to the depth coordinate, something like a function with options to select pointing upward or downward, and whether there is a deviation from vertical, etc. To be discussed!

emiliom, 2022-03-11:
Note: PR #577 was closed and replaced by #579

valentina-s, 2022-04-29:
I see the value in the generality of keeping the `echo_range` and `range_offset` separate. However, I would like to see how one can envision a whole pipeline for processing files. Many analysis/science operations will require only `depth, channel, ping_time` (it might be interesting though also to brainstorm the cases when people need the offset/tilt). So, in this case if people convert bunch of files do they store them in the general 'Sv' format or they calculate the depth and store them with a depth dimension? If they do the first, each time they access the file they will need to do this conversion operation to plot the data or do something else. It can be facilitated with a function, but then users will need to install `echopype` just for that function, while in general a lot of follow up analysis can happen on Sv `.nc` file just with `xarray`.  If they directly store it with the `depth` dimension then then store it in a "non-standard Sv" format? I wonder if a general `Sv` standard can allow for one or the other one, with offset/tilt stored to convert between them. So they both are considered "compliant".

leewujung, 2022-06-15:
@valentina-s and I have some conversation today about this. My thought was that we can provide functions to add `depth` as a data variable in the `Sv` dataset; similarly, I was thinking that for other needs such as interpolating the lat/lon positions from the `Platform` group to be along `ping_time` those become new data variables too in the `Sv` dataset. Users can just save a single copy of `Sv` dataset regardless of what operations have been done to it. BUT for this to work we should make the computation traceable by some chain of provenance which @emiliom and I briefly discussed some time ago during the v0.6.0 push.

Seems like time for a brainstorming session on how we want these functions to be.

Also tagging @b-reyes and @lsetiawan here so that you're aware of this discussion.

leewujung, 2022-06-21:
One more wrinkle to this is that in the current `echo_range` calculation the values below 0 are set to 0, which creates problems when we want to combined multiple datasets together (repeated values). In practice this does not matter in the use, but we need to fix this issue so that we can combine datasets easily.

leewujung, 2022-08-19:
A basic function for this functionality to take in depth_offset and tilt of the transducer is addressed by #738. Need to add functionalities to allow using data variables already stored in the EchoData object to add into the Sv dataset.

closed 2022-08-19 by leewujung