echostack-org/echopype #580: Remove NMEA sentence selection `open_raw`

leewujung, 2022-03-11:
## Context 

This is leftover work from the overhaul at v0.5.0:

We should remove the variable `NMEA_SENTENCE_DEFAULT` that is defined in 
https://github.com/OSOceanAcoustics/echopype/blob/5cf1153b3d292ce819276ed12c86816c694b75ec/echopype/convert/api.py#L26
and
https://github.com/OSOceanAcoustics/echopype/blob/5cf1153b3d292ce819276ed12c86816c694b75ec/echopype/convert/convert.py#L11  (The second one is not currently used -- we should deprecate the old convert mechanism, which will be addressed in #506.)

We are pretty clear now that we want the level 0 converted data product to be authentic reproduction of the raw data generated from the instruments. Therefore, there should not be any selection of data within `open_raw`, so we need to remove this mechanism.

We should have this mechanism as a data processing function for higher levels, since not all NMEA sentences have the same resolution and that can cause problems in plotting ship tracks, but that would be a separate issue/PR.

## Task
- [ ] Remove NMEA_SENTENCE_DEFAULT and related sentence selection mechanism
- [ ] Create test to ensure all sentences are stored in "/Platform/NMEA" -- one of the hake 2017 files would be good, because they have a mixture of NMEA sentence types

b-reyes, 2022-07-28:
@leewujung please correct me if I am misunderstanding something, but based off of [#673 (comment)](https://github.com/OSOceanAcoustics/echopype/issues/673#issuecomment-1130404416) it seems like we should not implement this (at least for the EchoData object). As I recall, it will cause several downstream issues that have little to no upside. 

> We are pretty clear now that we want the level 0 converted data product to be authentic reproduction of the raw data generated from the instruments. 

If our goal is to have level 0 be an authentic reproduction of the raw data, then does it make sense to make `open_raw` have an optional argument that would instead return a pandas dataframe that is just the parsed data?

leewujung, 2022-07-28:
@b-reyes : I am not sure what you mean, because this PR is exactly to REMOVE the _currently unused_ (and cannot be used) NMEA sentence selection mechanism, so that there is NO optional argument in `open_raw` to do any selection.

leewujung, 2022-07-28:
@b-reyes : reading what i wrote earlier again, I guess we were probably talking about different things... Maybe you were referring to the fact that we cannot preserve _all_ the NMEA sentences? If so, I agree.

I think my issue cobbled two things so probably can be revised. In my mind the thing to be removed is the option for user to say they only want a further subset of those GPS position sentences. (and that we still have to select these sentences out from all the NMEA sentences).

b-reyes, 2022-07-28:
> I think my issue cobbled two things so probably can be revised. In my mind the thing to be removed is the option for user to say they only want a further subset of those GPS position sentences. (and that we still have to select these sentences out from all the NMEA sentences).

@leewujung that makes sense now. Great, I have no problem with this. 

Does this new objective sound good @leewujung? 

Objective: Remove the option that allows the user to select a subset of the GPS position sentences and we should continue to select the same NMEA sentences. Specifically, we should only select the NMEA sentences: `["GGA", "GLL", "RMC"] `.

leewujung, 2022-07-28:
@b-reyes : Yes, I think you are spot on! Thanks for keeping clear eyes on this!  Feel free to edit the issue text as you see fit.