echostack-org/echopype #581: Separate out functions that bring data from processing level 0 to 1

leewujung, 2022-03-11:
## Context

From the workflow/pipeline perspective, we want to have clean distinctions between processing levels: 
- level 0 is data converted (losslessly) from raw instrument generated files
- level 1 is converted with other ancillary data combined with level 0 data.

Right now we have functions that brings data from level 0 to level 1 in `open_raw`, which blurs the line.

It seems that it would be better that we move those functions into methods of the EchoData object, like `echodata.update_platform()` where users can deliberately add additional info.

One thing I am not exactly sure about is what are the user-specified info/attributes/variables that should be part of `open_raw` so that appropriate metadata is included in the level 0 data product as per SONAR-netCDF4, and what are the other user-specified info/attributes/variables that should only be in level 1 data products.

The place where I saw this is in
- `_set_convert_params()` 
https://github.com/OSOceanAcoustics/echopype/blob/5cf1153b3d292ce819276ed12c86816c694b75ec/echopype/convert/api.py#L252-L254
where some user specified info is passed into `setgrouper` via this and make their way into outputs from `open_raw`
- the above also uses `_parse_NMEA()` (this is a bad name btw)
https://github.com/OSOceanAcoustics/echopype/blob/5cf1153b3d292ce819276ed12c86816c694b75ec/echopype/convert/set_groups_base.py#L146
to select the NMEA datagram to be stored under the group `/Platform/NMEA`.
The sentence-selection mechanism should be removed (see #580) from `open_raw`, but restored somewhere in the processing pipeline -- although the selection strictly speaking is a "minus" operation, and for level 0->1 it seems that "additions" are more appropriate.

Pinging @emiliom for inputs.

emiliom, 2022-03-16:
For reference: https://github.com/uw-echospace/data-processing-levels/blob/main/from_proposal.md

leewujung, 2022-12-01:
@emiliom : This issue seems pretty outdated and can be closed. I confirmed that:
- `_set_convert_params()` no longer uses `NMEA_SENTENCE_DEFAULT`
-  `_parse_NMEA` has been renamed to [`_extract_NMEA_latlon()`](https://github.com/OSOceanAcoustics/echopype/blob/b0b1490a9081400943ac2606e2e6fd2deb4ae71e/echopype/convert/set_groups_base.py#L161).

Just wanted to ping to confirm that we can close this.

emiliom, 2022-12-01:
I agree. Let's close this

closed 2022-12-01 by leewujung