echostack-org/echopype #587: Fix flake8 linting issue that causes all the PR's to fail

lsetiawan, 2022-03-15:
There seem to be flake8 issue in one of the file that is causing the pre-commit to trigger failure. See below.

[image]

leewujung, 2022-03-16:
I couldn't stand the X when fixing the indentation in #583 so just did this one.
I didn't look into if we can configure flake8 such that it skips strings automatically -- seems like it should be possible?  I just put in `#noqa` for these lines like all the other occurrences of the same error we have in the package. These long lines came from #574.

leewujung, 2022-03-23:
I'll close this since the immediate cause is resolved.

closed 2022-03-23 by leewujung