echostack-org/echopype #600: Update `typing_extensions` version in requirements

leewujung, 2022-03-28:
Right now conda does not support the version specifier we have in `requirements.txt`
``` 
typing-extensions~=3.10
```

Changing it in 2 ways results in different behavior (as per [the pip example here](https://pip.pypa.io/en/stable/reference/requirements-file-format/#example)):
- With `typing-extensions>=3.10` I got version `4.1.1`
- With `typing-extensions==3.*` I got version `3.10.0.2`

We can fix this by using both `>=3.10` and `==3.*`, but I think we should test if `typing-extensions=4.X` works so we can keep up with updates of dependencies.

lsetiawan, 2022-03-28:
With PR #604 on the way. When echopype minimum is py 3.8, we will not need this anymore.

emiliom, 2022-04-06:
I think we can close this issue now, after #599 and #604

leewujung, 2022-04-06:
Agreed. :)

closed 2022-04-06 by leewujung