echostack-org/echopype #606: Backward/forward compatibility wrt v0.6.0 group structure/variable name change

leewujung, 2022-03-30:
## Context 
We are coming up against multiple planned breaking changes in v0.6.0. The changes include netCDF group structures as well as coordinates and variable names and attributes.

We know that many users, including both OOI and NCEI have converted some non-trivial volume of data into zarr or nc using echopype v0.5.x, so it is imperative that all processing can accommodate these already converted data.

In light of the group structure changes, we also want to ease the transition for ourselves such that we do not disturb the entire codebase significantly all at once (so that we can focus on convention compliance for v0.6.0).

## Approach
There are two types of changes we need to deal with:
1. group structure changes
2. variable name changes

1 is more involved, and 2 is a simple variable name change once data are read into xarray datasets, so below is mainly about 1.

The `EchoData` object is an encapsulated form of the data and metadata. Right now we have `EchoData` class attributes that correspond with the netCDF groups. For example:
- `EchoData.sonar` is an xarray dataset from the `root/Sonar` group
- `EchoData.beam` is an xarray dataset from the v0.5.x `root/Beam` group
- `EchoData.nmea` is an xarray dataset from the v0.5.x `root/Platform/NMEA` group

In v0.6.0, 

, so the approach we can take here is to under the hood hook up the old groups/variables in v0.5.x files to the right places with the right names in the v0.6.0 `EchoData` object. This way we can change all the downstream code to use the new (v0.6.0) `EchoData` groups/variables while still support files converted with 0.5.x.

## Tasks
- [ ] connect v0.5.x `root/Beam` group to v0.6.0 `EchoData["root/Sonar/Beam_group1`

b-reyes, 2022-03-30:
Thank you @leewujung for putting this together! I have been working on this conversion problem, so I am happy to work on several of these items. I think that once we establish the sensor-specific files in #596, then we can proceed with making the code forward compatible. The following is the first step I would take (input is welcome) to address this issue. Note that this is primarily focused on changes to downstream items related to `open_converted`.
* Find the version of echopype that created the zarr/nc file and use the sensor-specific files created in #596 to open and read in the file into a EchoData object. 
    * This will require changes to [`__load_file`](https://github.com/OSOceanAcoustics/echopype/blob/f0ce6b6a4acc61a92693fa001619744dfac5a8a5/echopype/echodata/echodata.py#L603). 
        * Replace `self.group_map.items()` with a sensor-specific dict obtained from #596. 
        * Replace [lines 616-620](https://github.com/OSOceanAcoustics/echopype/blob/f0ce6b6a4acc61a92693fa001619744dfac5a8a5/echopype/echodata/echodata.py#L616-L620) with a function that correctly assigns the group based on the sensor-specific file. 
        * Add a depreciation warning/prompt to save to zarr/nc, if conversion is necessary
        * Apply change to variable names, if necessary

From my understanding, if we end up implementing datatree, we would only need to modify the above items slightly.

emiliom, 2022-03-30:
Wow, thank you @leewujung !! This is really helpful.

> Group structure changes

I think what you describe as *forward compatibility* is what we had agreed on. That is, to temporarily retain (at least in 0.6.0) the current `EchoData` group referencing (`.beam`, `.sonar`, etc) but map them to the 0.6.0 group paths (ie, correct SONAR-netCDF4 v1 paths). 

When using `open_converted` to read a 0.5.x file (which would use [open_datatree](https://xarray-datatree.readthedocs.io/en/latest/generated/datatree.open_datatree.html) behind the scenes), the group paths and names would be "wrong" (ie, 0.5.x paths) vis a vis 0.6.x. As @b-reyes discussed above (thanks for the additional details, @b-reyes!), the group mappings we're discussing in #596 (and #598) are intended to address this challenge.

I think there are still challenges we need to work through, and let's discuss them at our meeting tomorrow. For example, all these group mappings and variable renamings (and other changes -- see below) have a consequence I hadn't thought about: `open_converted` applied on a 0.5.x file might not be doable with lazy loading alone.

> There are two types of changes we need to deal with:

Actually, there's a third type that's related to "variable name changes" but a lot more consequential: changes to a dimension/coordinate that go beyond just a name change. Specifically:
- Addition of the new `beam` dimension (#520). EK80 is the only exception, where the change really is just a simple name change to the existing `quadrant` dimension.
- Fundamental change to the meaning of the `frequency` dimension (#566). It's not just a simple renaming to `channel`. The frequency information will now be stored somewhere else, in the new `frequency_nominal(channel)` variable. Code that will need to be changed includes query patterns that relied on the `frequency` coordinate, and `open_raw` operations that populate the frequency information.

leewujung, 2022-03-30:
> all these group mappings and variable renamings (and other changes -- see below) have a consequence I hadn't thought about: open_converted applied on a 0.5.x file might not be doable with lazy loading alone.

I am not sure "what it takes" to rename the coordinate variables (i.e. does the renaming operation requires actually loading data into memory?), but this part to hook up the v0.6.0 groups and subgroups to our current flattened `EchoData.GROUP` as part of `open_converted` should be fine with lazy loading.

For the third type of changes -- Thanks! I agree this is a different type that should be listed to the main issue description! I'll add it later today.
> - Addition of the new beam dimension (https://github.com/OSOceanAcoustics/echopype/issues/520). EK80 is the only exception, where the change really is just a simple name change to the existing quadrant dimension.

This is true, but I think the processing of the other sonar models can be achieve without much disturbance of the code by using `squeeze` to remove that extra dimension `beam` with dim=1.

> Fundamental change to the meaning of the frequency dimension (https://github.com/OSOceanAcoustics/echopype/issues/566). It's not just a simple renaming to channel. The frequency information will now [...]

Agreed. I think this will be a tedious PR thought I don't see immediately major obstacles.

emiliom, 2022-03-30:
> but this part to hook up the v0.6.0 groups and subgroups to our current flattened EchoData.GROUP as part of open_converted should be fine with lazy loading.

100% correct. I'm glad you pointed it out, since it's easy (for me at least) to get lost in the details.

> This is true, but I think the processing of the other sonar models can be achieve without much disturbance of the code by using squeeze to remove that extra dimension beam with dim=1.

Oh, that's brilliant!!! I hadn't thought of that. Hopefully it will pan out that way.

b-reyes, 2022-03-30:
> Actually, there's a third type that's related to "variable name changes" but a lot more consequential: changes to a dimension/coordinate that go beyond just a name change. Specifically:

Thank you for pointing this out @emiliom, I will be sure to keep this in mind. Perhaps I can lay the groundwork for this forward compatibility by just implementing the restructuring of the groups and changing `range_bin` to `range_sample`, then we can work together to see the best way to go about the change to `channel`.

leewujung, 2022-03-30:
> just implementing the restructuring of the groups and changing range_bin to range_sample

That will be great! This will also let us know if renaming a coordinate requires loading the coordinate data into memory. But I think even if it is, the penalty may not be a problem for the size of data files we are talking about at the moment -- until we fix the problem with `combine_echodata` to produce a gigantic zarr file (since right now people run out of memory to combine too many files 😅, so that difficult scenario to have to rename the coordinate from a gigantic dataset actually does not happen -- one of the times where we have the blessing of missing functionalities...?).

emiliom, 2022-03-31:
> Perhaps I can lay the groundwork for this forward compatibility by just implementing the restructuring of the groups and changing `range_bin` to `range_sample` then we can work together to see the best way to go about the change to `channel`

That sounds great, @b-reyes ! Specially the range_bin > range_sample conversion (in `open_converted`, right?). For the restructuring of the groups, once we settle down on the form of the mapping dict (https://github.com/OSOceanAcoustics/echopype/issues/596#issuecomment-1082471372), you can start working on it. But bear in mind that @lsetiawan [said](https://github.com/OSOceanAcoustics/echopype/issues/567#issuecomment-1082324935) he'll work on the EchoData.grouplabel <> EchoData.getter["group-path"] mappings on Friday. Your work on `open_converted` to map groups from 0.5.x to 0.6.x will probably have to interact with Don's work; though I don't have a good feel for how much of a handshake there needs to be.

leewujung, 2022-03-31:
> But bear in mind that @lsetiawan https://github.com/OSOceanAcoustics/echopype/issues/567#issuecomment-1082324935 he'll work on the EchoData.grouplabel <> EchoData.getter["group-path"] mappings on Friday. Your work on open_converted to map groups from 0.5.x to 0.6.x will probably have to interact with Don's work; though I don't have a good feel for how much of a handshake there needs to be.

I suggest that we talk this through and agree on who will do what in the call tomorrow morning. There are 2 steps to consider:

1. `open_tree` can provide the mechanism for linking both the v0.5.x and v0.6.x files to the current flattened `EchoData.GROUP` structure _without_ us changing much in the downstream computations (other than coordinate and variable name fixes).

2. To actually enable the EchoData.grouplabel <> EchoData.getter["group-path"] mapping. Directly using `open_tree` wouldn't work, since `EchoData["Sonar/Beam_groupX"]` only works for v0.6.x and `EchoData["Beam"]` only works for v0.5.x, so something has to happen there.

b-reyes, 2022-03-31:
@emiliom yes, once we settle #596 I will start working on this, I plan to follow the outline I created above. I will be sure to discuss this with @lsetiawan in the meeting today.

leewujung, 2022-05-27:
Done!
Remaining things to be picked up include:
- https://github.com/OSOceanAcoustics/echopype/pull/685#discussion_r879016132
- #708

closed 2022-05-27 by leewujung