echostack-org/echopype #618: Add echopype software and version provenance attributes to Sv and MVBS datasets

emiliom, 2022-04-09:
An easy first step would be to adapt the SONAR-netCDF4 v1 `Provenance` attributes, to be inserted into Sv and MVBS datasets as global attributes. Here's a sample. Instead of "calibration", the prefix might be "calibration" or something more generic ("processing").

From the `Provenance` group of a converted file:
```
conversion_software_name : echopype
conversion_software_version : 0.5.4
conversion_time : 2021-10-29T00:20:18Z
src_filenames : s3://ncei-wcsd-archive/data/raw/Bell_M._Shimada/SH1707/EK60/Summer2017-D20170728-T195219.raw
```

Should this go into version 0.6.0?

leewujung, 2022-04-09:
> Instead of "calibration", the prefix might be "calibration" or something more generic ("processing").

Somehow "calibration_software_name" feels a bit weird, "processing_software_name" seems more natural?

emiliom, 2022-04-09:
> Somehow "calibration_software_name" feels a bit weird, "processing_software_name" seems more natural?

I don't know about the weirdness (I was going off the fact that echopype calls the creation of Sv a "calibration"). But I like "processing" better b/c it's more reusable and we could use the attribute names for both Sv and MVBS.

leewujung, 2022-04-10:
Ok, let's go with using "processing_*" for these attributes then. This is also something that higher processing levels can use.

emiliom, 2022-04-10:
Just realized that this issue is related to a pretty old issue, #221. Maybe we could fold the `remove_noise` aspect of #221 into this one, then close it?

leewujung, 2022-04-11:
> Maybe we could fold the remove_noise aspect of https://github.com/OSOceanAcoustics/echopype/issues/221 into this one, then close it?

Sounds good!

emiliom, 2022-05-05:
#621 closes this issue

closed 2022-05-05 by emiliom