echostack-org/echopype #620: Change `src_filenames` string attribute to `source_filenames` list-of-strings variable

emiliom, 2022-04-09:
This kept falling through the cracks. echopype uses `src_filenames` (in the `Provenance` group), and sets it as a string when it's a single file name (ie, from a simple, single raw file conversion). But in the convention its name is `source_filenames` and it's actually defined as a *variable*, not a global attribute. Also, it's a "vector of datafile names that were used ...", not a single string.

In echopype  `src_filenames` is used only in just a couple of places at this time. BTW, one of those places, in `SetGroupsEK60.set_provenance`, seems overly redundant of `SetGroupsBase.set_provenance`

leewujung, 2022-04-10:
> in the convention its name is source_filenames and it's actually defined as a variable, not a global attribute. Also, it's a "vector of datafile names that were used ...", not a single string.

This is great, thanks for discovering this.
In `combine_echodata` @imranmaj @lsetiawan and I actually arrived at the conclusion to have `src_filenames` as a data variable, and as a vector also. 
https://github.com/OSOceanAcoustics/echopype/blob/185cc1e6f444538a70996578b5006047e2a51565/echopype/echodata/combine.py#L27-L39
so we can now standardize this according to the convention! :D

emiliom, 2022-05-12:
Closed by #674

PR #674 (2022-05-12): Change src_filenames string attribute to source_filenames list-of-strings variable
Closes #620. Implemented using a new prov.py function, `source_files_vars`, that is reused wherever the `source_filenames` variable is needed, including in `assemble_combined_provenance` where it already existed. Addresses convention adherence.
closed 2022-05-12 by emiliom