echostack-org/echopype #624: Possible required change in `Sonar` group description

b-reyes, 2022-04-12:
In the current development version, the `Sonar` group has the description "contains specific metadata for the sonar system." 

However, the `Sonar` group now contains more than metadata, it now contains the beam groups. Should we change the description to take this into account? For example, the description could be something along the lines of "contains the metadata and beam groups of the sonar system."

emiliom, 2022-04-13:
That makes sense. Looking at the group description used in the convention document, I would modify the description you're suggesting to, say: "contains sonar system metadata and sonar data beam groups"

b-reyes, 2022-04-13:
Sounds good. Thank you @emiliom for your suggestion. I will go ahead and make this change.

emiliom, 2022-05-07:
Addressed in #667

closed 2022-05-07 by emiliom