echostack-org/echopype #634: Beam group variable suffixes `_alongship/athwart` vs  `_major/minor`

b-reyes, 2022-04-19:
In issue #520 it was found that variables within the Beam group with the suffixes `_alongship/athwart` do not appear in the convention. It seems that their original intent was possibly to coincide with the variables `beamwidth_receive/transmit_major/minor` in the convention. Based on the convention beamwidth_receive_major/minor are listed as "M" and beamwidth_transmit_major/minor are listed as "MA" (mandatory when applicable).  Thus, this conflict with the convention needs to be addressed. In #520 @leewujung made the following comment in regards to the matter: 

> For the athwartship/alongship vs major/minor question:
> Thinking out loud here, it seems that we can do the following:
> 
> 1. use what @gavinmacaulay pointed out in v2 in our compliance in v1 and describe clearly in the `comment` attribute [of the `_alongship/athwart` variables what our interpretation of these variables are] and reference the sources.
> 2. set `*_major/minor` to NaN and use `*_athwartship/alongship` in the variables as in the raw data, and describe clearly that this is a decision we've made in echopype.
> 
> I am inclined slightly to option 2 so that there is no ambiguity in interpreting the variables on what they are. When we need to go to v2, connecting the right variables to according the v2 definitions is straightforward.

leewujung, 2022-04-19:
Hey @b-reyes : what I meant was that I will take the part about the 1-way and 2-way beamwidth business. For the overall handling I think you'll be better suited since you're already working on #520 .

emiliom, 2022-04-20:
> In issue #520 it was found that variables within the Beam group with the suffixes `_alongship/athwartship` do not appear in the convention. It seems that their original intent was possibly to coincide with the variables `beamwidth_receive/transmit_major/minor` in the convention.

To add to this: there is no doubt about this original intent! Otherwise, @leewujung can confirm :stuck_out_tongue_winking_eye: 

For echopype 0.6.0 we also agreed to add a `comment` attribute to each `_alongship/athwart` variable that is based on a convention variable, indicating the variable it broadly corresponds to. It could be something like this:
```
beamwidth_receive_athwartship
:comment: "Corresponds to beamwidth_receive_major in SONAR-netCDF4 vers 1, but introduced in echopype for echosounders"
```

The change @leewujung describes above will also be for 0.6.0. It sounds like it's fairly easy.

We leave for a later time the larger issues of how/if these variables can be made to be coherent with corresponding v1 convention variables, and how they correspond to new variables in v2 for split-beam sonars (eg, [single_target_athwartship_angle](https://raw.githack.com/ices-publications/SONAR-netCDF4/master/Formatted_docs/crr341.html#_proposed_single_target_additions)). We note also that the `_alongship/athwartship` descriptors don't make clear sense for echosounder deployments in moorings.

leewujung, 2022-05-05:
@leewujung will work on a PR to:
- implement the suggested variable attribute
- take care of the 1-way vs 2-way beam pattern issue exposed in #520

emiliom, 2022-05-18:
I think we can close this issue, right?

leewujung, 2022-05-18:
Agreed! I'll close this now.

closed 2022-05-18 by leewujung