echostack-org/echopype #642: Move some AZFP Beam variables to Vendor and Platform groups

emiliom, 2022-04-21:
Following up from https://github.com/OSOceanAcoustics/echopype/issues/520#issuecomment-1104626403 and Slack exchange about `tilt*` variables.

**1. Move the following AZFP variables from `Beam_group1` to other groups.**

- To `Platform` group:
  - `tilt_x/y(ping_time)`. Note that this is a *processed* variable. Eventually (later), add helpful attributes if it doesn't already have them. eg, the orientation of the x and y axes.
- To `Vendor` group: 
  - the variables below are specific to AZFP as engineering outputs, some can be converted to "real" data variables in the Environment or Platform groups
    - `temperature_counts(ping_time)`
    - `tilt_x/y_count(ping_time)`
  - the variables below are specific to AZFP and used for AZFP-specific calibration
    - `DS(channel)`
    - `EL(channel)`
    - `​​TVR(channel)`
    - `VTX(channel)`
    - `Sv_offset(channel)`
    - `number_of_samples_digitized_per_pings(channel)`
    - `number_of_digitized_samples_averaged_per_pings(channel)`

**Eliminate the variable `cos_tilt_mag(ping_time)`.** It's redundant since it can be [easily calculated](https://github.com/OSOceanAcoustics/echopype/blob/main/echopype/convert/parse_azfp.py#L180) from `tilt_x/y`. It's also not used anywhere in echopype.

b-reyes, 2022-05-06:
@emiliom since we are moving several variables out of `Beam_group1`, what should we do with their associated attributes? Specifically: 

- We are moving `tilt_x/y(ping_time)` to the `Platform` group and `tilt_x/y_count(ping_time)` to the `Vendor` group. So, what should we do with the attributes `tilt_X_a/b/c/d` and `tilt_Y_a/b/c/d`? Should these be moved to the `Platform` group too?
- `temperature_counts(ping_time)` is being moved to the `Vendor` group. What should we do with the attributes `temperature_ka/kb/kc/A/B/C`? 
- What should be done with the attributes: `number_of_frequency`, `number_of_pings_per_burst`, `average_burst_pings_flag`?
- I am guessing that `conversion_equation_t` and `beam_mode` can stay in `Beam_group1`?

leewujung, 2022-05-06:
> * We are moving `tilt_x/y(ping_time)` to the `Platform` group and `tilt_x/y_count(ping_time)` to the `Vendor` group. So, what should we do with the attributes `tilt_X_a/b/c/d` and `tilt_Y_a/b/c/d`? Should these be moved to the `Platform` group too?

Please move `tilt_X_a/b/c/d` and `tilt_Y_a/b/c/d` to the `Vendor` group. These are coefficients that are used to convert the `*_counts` to the actual values.

> * `temperature_counts(ping_time)` is being moved to the `Vendor` group. What should we do with the attributes `temperature_ka/kb/kc/A/B/C`?

Please move `temperature_ka/kb/kc/A/B/C` to the `Vendor` group also. These are coefficients that are used to convert the `*_counts` to the actual values.

> * What should be done with the attributes: `number_of_frequency`, `number_of_pings_per_burst`, `average_burst_pings_flag`?

These should also go to `Vendor`, since there does not seem to be an obvious reason why they need to be in `Beam_group1`, and they are vendor-specific.

> * I am guessing that `conversion_equation_t` and `beam_mode` can stay in `Beam_group1`?

These are convention defined variables, so I think should stay.

For the coefficients, what I don't know is if they should remain attributes or should be variables. @emiliom would have better insights here.

emiliom, 2022-05-08:
> For the coefficients, what I don't know is if they should remain attributes or should be variables. @emiliom would have better insights here.

In general, anything going into `Vendor` can be structured any way we want. I would keep them as is (no change from attributes to variables) for 0.6.0, unless there are very specific, compelling reasons to change them.

b-reyes, 2022-05-09:
> In general, anything going into `Vendor` can be structured any way we want. I would keep them as is (no change from attributes to variables) for 0.6.0, unless there are very specific, compelling [...]

@emiliom that makes sense to me. @leewujung do you have any reasons in mind as to how these coefficients might benefit from being turned into variables?

leewujung, 2022-05-09:
Not really, I think we can keep them as attributes.

b-reyes, 2022-05-10:
The items in this issue were addressed via PR #669.

closed 2022-05-10 by b-reyes