echostack-org/echopype #644: Squeeze out `beam` dimension in `compute_Sv/TS` outputs

leewujung, 2022-04-22:
With the addition of the `beam` dimension to a large number of variables in the `Beam_groupX` groups for compliance with the convention, now the outputs of `compute_Sv/TS` would also have a `beam` dimension. This dimension is of length=1 for AZFP, EK60 and EK80 power/angle data, but of length=4 for EK80 complex data.

The EK80 length=4 output does not make sense, since these are from a single split-beam transducers. There is a conceptual incongruence and potential confusion that takes place here. This is related to how we use the beam dimension to store data from each element of a split-beam transducer. (See https://github.com/OSOceanAcoustics/echopype/pull/638#issuecomment-1105807694 for detail)

The consensus was to squeeze out this extra `beam` dimension at the outputs of `computer_Sv/TS`.

We need to decide on our approach:
- option 1: modify the code so that the redundant `beam` dimension is dropped so that the variables used in the computation "makes sense"
- option 2: just carry through the redundant `beam` dimension toward the end of the computation and drop before returning the outputs.

Having done some of option 1 (those are "saved" in a local branch now), option 2 seems a cleaner approach. However it probably would have an impact computationally. 🤔

emiliom, 2022-04-22:
Hmm. I don't think I can add anything w/o evaluating the code. It sounds like it's a decision where the down sides are code complexity (option 1) vs computational performance (option 2)? How about if you push your local branch to your fork, and point us to the code you've changed?

For reference, the test @b-reyes reported in https://github.com/OSOceanAcoustics/echopype/pull/638#issuecomment-1106744630 https://github.com/OSOceanAcoustics/echopype/pull/638#issuecomment-1106744630 is [tests/calibrate/test_calibrate.py::test_compute_Sv_ek80_matlab](https://github.com/OSOceanAcoustics/echopype/blob/dev/echopype/tests/calibrate/test_calibrate.py#L176)

emiliom, 2022-05-18:
What did we decide about this, for v0.6.0?

leewujung, 2022-05-18:
Ugh, forgot to come back to this. I _think_ we decided to go with option 2, but it looks like that I have an action item to push up some option 1 attempt. Let me do that shortly. But looking at this again, it seems that just squeezing out that dimension at the end is more convenient. Once we get through v0.6.0 we can then work on the computational efficiency part.

emiliom, 2022-05-20:
We can close this now.

PR #687 (2022-05-20): Squeeze and drop the redundant `beam` dimension in calibration outputs
This PR closes #644.

I'm taking the approach to just squeeze and drop the redundant `beam` dimension at the end of the calibration operations. There may be efficiency advantage to take care of this extra dimension earlier in the computation, but let's delay this part to post-v0.6.0.

I've also noted the variables that have the redundant `beam` dimension (as comments) as I go through to make this change.

## Tasks
- [x] Squeeze out `beam` from `CalibrateEK._cal_power()`
- [x] Squeeze out `beam` from `CalibrateEK._cal_comple [... cut]
closed 2022-05-27 by leewujung