echostack-org/echopype #659: Improve parsing config XML in AZFP files

leewujung, 2022-05-03:
Currently only a subset of variables are parsed from the AZFP XML files. These are the ones that were parsed in the reference Matlab code from the manufacturer ASL Env Sci.

We recently found that in order to encode a unique `channel` identifier, it would be the best to use the field `Frequency Number` in the AZFP XML file. However, the current simplistic XML parsing function does not support that. Let's take this chance to improve the robustness of this XML parsing, and also to include the variable units which were not included in the current parser.

For needing a unique `channel` identifier, see #566, which is addressed by #657.

This is similar to #468.

## Tasks
- [ ] improve robustness of AZFP XML parser, using something like `lxml`
- [ ] improve AZFP XML parser to include all config variables
- [ ] include variable units in the XML file to parser and store accordingly in the set_groups methods

emiliom, 2023-09-18:
The first two tasks are addressed in #1151.

I think the third task, "include variable units in the XML file to parser and store accordingly in the set_groups methods", was partly addressed in v0.8.0/0.8.1. Other aspects related to it will be addressed via #1171.

praneethratna, 2023-10-14:
Closing, since this was fixed in #1182

closed 2023-10-14 by praneethratna

emiliom, 2023-10-15:
I've checked off the third task (units), since it was still unchecked. See my note above, https://github.com/OSOceanAcoustics/echopype/issues/659#issuecomment-1724577701

PR #1135 (2023-09-18): Changed parser in `parse_azfp.py` to maintain consistency over other parsers.
Fixes #659. Now `parse_azfp.py` uses `xml.etree.ElementTree` instead of `xm.dom.minidom` for parsing.

Update: More parameters are being parsed now, and some of those were added to the EchoData object via set_groups.

CC @leewujung @emiliom