echostack-org/echopype #677: `xml_path` is not contained in `source_filenames` for the AZFP sensor

b-reyes, 2022-05-12:
While looking into PR #674, it was found that currently we only store the `raw_file` path from `open_raw` as an attribute (after PR #674 this will become the variable `ed.provenance.source_filenames`). However, for AZFP we have both the `raw_file` and `xml_path` as inputs and the file names for these can be quite different. After discussing this with @emiliom, it seems like we should make it so that we store both the `raw_file` and `xml_path` in the variable `ed.provenance.source_filenames` or modify this variable in someway for the AZFP sensor.

emiliom, 2022-09-09:
I only see three options here, and none of them is great. For the case of applying `open_raw` on a single AZFP file:

1. Rather than than entering a single string for the data file (the file path, eg, "/my/path/17082117.01A"), enter a string array with both file paths (eg, `["17082117.01A", "17041823.XML"]`)
2. Introduce a new variable, say, `sourcemeta_filenames`, to hold the metadata/XML: "17041823.XML". Then `source_filenames` will remain as is, listing only the data file: "17082117.01A"
3. Ignore the metadata/XML file path

1 is simple to implement and doesn't introduce new variables. It passes the complexity down to the `combine_echodata` operation, which will need to figure out how to handles pairs of paths that represent a single data source. It could use string representations of string tuples as individual array elements, eg, `['("17082117.01A", "17041823.XML")', '("27082117.01A", "27041823.XML")', ...]`). But then that's not what the convention says, and it'll require special logic for echopype to be able to detect and handle both this form and the current, simple string array cases like EK60, eg, `["file1.raw", "file2.raw", ...]`.

2 is clean, but it introduces a new variable not in the convention.

3 is easy: do nothing! :wink:

@leewujung, thoughts?

leewujung, 2022-09-09:
I vote for 2. I think it is ok to have a new variable as long as it is clear what that is. I also like the part that it makes the core logic in `combine_echodata` intact. 

For `sourcemeta_filenames`: in `combine_echodata` I suggest we use the same logic as `source_filenames` to store the combined outcome (whatever form that takes as part of the `source_filename` clean-up).

b-reyes, 2022-09-09:
@emiliom thank you for looking into this. From the options you have proposed, option 2 does seem to be the best solution. 

For option 2, will the variable `sourcemeta_filenames` exist for the other sensors too? If so, what data will it hold? If this variable only exists for the AZFP sensor, then it might make more sense to have a more specific name for it. Maybe `source_xml_filenames`?

emiliom, 2022-09-13:
Thanks to both! I'll implement this ("option 2"). 

As for the name of this new variable, I proposed `sourcemeta_filenames` as something generic not tied to one specific file format (XML) so it can be reused with any potential -- but unspecified! -- future instruments that also use an external metadata file. Right now this is hypothetical, since AZFP is the only such implementation we have and it's unlikely we'll add others in the near future. 

A generic name is also consistent with `source_filenames` itself, which is not specific to a file format. A slightly more readable form could be `source_meta_filenames`. Instead of "meta" we could also come up with another generic and readily understood word, like "instrument". We could also switch it around like `meta_source_filenames`.

For the other instruments echopype supports, or at least the Simrad instruments, there is no external metadata/instrument file. We could choose to always include this new variable and populate it with nulls, regardless of instrument. Or include it only if it's used.

leewujung, 2022-09-13:
Somehow l like `meta_source_filenames` better, but I am fine either way. 

I like the option of only include this variable only if it is used....

b-reyes, 2022-09-13:
@emiliom I am fine with the names you have proposed based on your explanation. I would suggest that we add some attribute that explains the contents though? Maybe we can have some variation of the following comment attribute:

```
meta_source_filenames.attrs["comment"] = Path names corresponding to additional data sources
``` 
or
```
meta_source_filenames.attrs["comment"] = Path names corresponding to the additional XML data source
``` 
I would like this to be more specific (the second option) and change with regards to the instrument, but that seems to go against what you said above.

> I like the option of only include this variable only if it is used....

I agree with @leewujung I think we should only include the variable if it is used.

emiliom, 2022-09-14:
From #806: Change `filenames.long_name` to something like "Index for data source file names", to ensure `long_name` is no longer exclusively tied to `source_filenames`

emiliom, 2022-09-22:
In PR #814 there is a helpful discussion about downstream propagation of the path(s) coming from `xml_path`, in `combine_echodata` and calibration functions. Those tasks will not be handled in the upcoming release (0.6.3), but we'll come back to them afterwards.

leewujung, 2022-09-26:
@b-reyes and @leewujung decided to merge #814 first without this last comment https://github.com/OSOceanAcoustics/echopype/pull/814#pullrequestreview-1120520503 is addressed. Noting it here so that it can get picked up later.

emiliom, 2022-09-29:
Here's the comment @leewujung referred to:

> @emiliom : thanks for the PR! I have a minor comment that does not require action, I was just wondering what you meant.
>
> I have an overarching question on the function design:
`utils/prov::source_files_vars` returns 3 single-element dicts with keys almost identical to the names of the dicts, and values being tuples that can be passed in to construct xarray dataArrays. Why not just a single dict with those same key-value pairs? The downstream code don't seem like would be impacted much.

I don't remember, off the top of my head. Probably inertia. As far as I can tell it could easily be replaced with a single dict, and I agree that'd make the code cleaner.

leewujung, 2022-10-13:
The core issue of this is address, but we need that final change to make the code cleaner. @emiliom : do you want to do that now before we forget?

emiliom, 2022-10-13:
Yeah, I'll do this by tomorrow

emiliom, 2022-10-15:
Completed in PR #852. Closing

closed 2022-10-15 by emiliom