echostack-org/echopype #702: Reduced sampling rate Power/Angle samples

marianpena, 2022-05-23:
Hi, 
I've been testing echopype with EK80 data recorded in CW mode and reduced sampling rate Power/Angle samples; backscatter looks fine but Sv is full of NaN values. I used ep.calibrate.compute_Sv(ed,waveform_mode="CW",encode_mode='power')

Echopype specifies 2 modes ("complex"` and `"power") but three data formats are possible:

The Stored sampled data for WBTs running CW options can be used to reduce the amount of recorded data when you are using a Wide Band Transceiver (WBT) with CW pulses.
• 	Select Complex samples to use the default data format.
• 	Select Power/Angle to reduce the file sizes.
• 	Select Reduced sampling rate Power/Angle samples to minimize the file sizes.

Is this last one supported?

Thanks!

marianpena, 2022-05-23:
Ok I found the error and how to fix it. Temperature, salinity and depth had the same length as ping_time but only the first value was filled (the rest as NaN). I fixed it by specifying:
env_params = {
    'temperature': ed.environment.temperature[0].values,   # temperature in degree Celsius
    'salinity': ed.environment.salinity[0].values,     # salinity in PSU
    'pressure': ed.environment.depth[0].values,     # pressure in dbar
}
and then ep.calibrate.compute_Sv(ed,waveform_mode="CW",encode_mode=encode_mode, env_params=env_params)

leewujung, 2023-03-09:
Hey @marianpena : I have been doing some refactoring of calibration code, which contains particularly parts about `cal_params` and `env_params` intake. I think my changes fixed the issues you exposed. Could you point me a sample file and a snippet of code to test? I don't have access to an EK80 system right now to produce reduced sampling rate data. Thanks!