echostack-org/echopype #704: Review `concat_dims` in `combine_echodata`

b-reyes, 2022-05-23:
In `combine_echodata` for some groups we use the function `xr.combine_nested`, which uses `concat_dims` from `echopype/core.py`. As of right now, it appears that we are always using the `default` key in `concat_dims` as the input to `xr.combine_nested`. I think this should be investigated further. I believe we should be using the key that is associated with the group and not always the `default` key.

leewujung, 2022-10-13:
@b-reyes : I believe you have already address this in the recent combine_echodata PRs

b-reyes, 2022-10-13:
@leewujung yes, I do address this in my recent `combine_echodata` PRs. The key `concat_dims` in `SONAR_MODELS` within `echopype/core.py` is no longer used. In fact, we should remove this. I can create a PR to do this right now.

b-reyes, 2022-10-13:
PR #849 removed `concat_dims` and `concat_data_vars` from `SONAR_MODELS`. For this reason, this issue is no longer relevant. I will go ahead and close it.

closed 2022-10-13 by b-reyes