echostack-org/echopype #722: Avout parsing message from open_raw

cornejotux, 2022-06-13:
This is more a question than an issue.
Is there any way to avoid getting the message "15:32:08  parsing file XXX_T165053.raw, time of first ping: 2022-May-14 16:50:53" when reading an EK80 raw file with open_raw?

I'm loading a few hundred files with a loop and instead of the message I want to get only a progress bar, which right now is re-display every time a file is loaded.

This is the code I'm using:

rawFiles contain all files to be read.

`ed =open_raw(rawFiles[0], sonar_model='EK80')
gps = ed['Platform'].latitude.to_dataframe().join(ed['Platform'].longitude.to_dataframe())

from tqdm import tqdm
with tqdm(total=len(rawFiles)) as pbar:
    for i in rawFiles[1:(len(rawFiles)-1)]:
        ed = open_raw(i, sonar_model='EK80')
        gps_df = ed['Platform'].latitude.to_dataframe().join(ed['Platform'].longitude.to_dataframe())
        frames = [gps, gps_df]
        gps = pd.concat(frames)
        pbar.update(1)`

Thanks in advance

leewujung, 2022-06-14:
@cornejotux : We currently don't have this implemented but it has been requested in another issue #457. Would you be able to give it a shot? It'd be great to use the python logging functionality to do this and add a flag to let user control whether or not to print to screen or just save to a log file.

emiliom, 2022-06-14:
As @leewujung pointed out, this issue is a duplicate of #457. I suggest we close it and move the discussions there.

cornejotux, 2022-06-14:
@leewujung not sure if I'm up to the task but I would take a look and see if I can contribute somehow.
Actually, I would love to.

leewujung, 2022-06-14:
Thanks @cornejotux !

emiliom, 2022-06-19:
Closing this issue to shift the discussions on this topic to #457. Thanks again @cornejotux ! Looking forward to your contribution on this issue

closed 2022-06-19 by emiliom