echostack-org/echopype #726: Add function to regrid Sv data

leewujung, 2022-06-15:
It is a pretty common need to regrid Sv data, since different channels (frequencies) may be configured differently to collect data with different `sample_interval` even if the total range collected is the same. 

Proposal: add this to the `preprocess` subpackage.

leewujung, 2023-09-13:
@leewujung will add a link here to a ref implementation from IMR.

lsetiawan, 2023-09-13:
This should be optional... potentially use: https://xesmf.readthedocs.io/en/latest/index.html

leewujung, 2023-09-27:
I found my notes from before. Here are two resources, one from IMR and the other pyEcholab.

IMR:
- it seems that `_resampleWeight`, `_regrid`, and `regrid_sv` are the functions used by IMR [here](https://github.com/CRIMAC-WP4-Machine-learning/CRIMAC-preprocessing/blob/master/CRIMAC_preprocess.py#L355-L463).
- it also seems that the ditched the resampleWeight thing in [this line](https://github.com/CRIMAC-WP4-Machine-learning/CRIMAC-preprocessing/blob/master/CRIMAC_preprocess.py#L455) under `regrid_Sv`

pyEcholab:
- [grid.py](https://github.com/CI-CMG/pyEcholab/blob/RHT-EK80-Svf/echolab2/processing/grid.py) seems to be generating just the grid boundaries
- [integration.py](https://github.com/CI-CMG/pyEcholab/blob/RHT-EK80-Svf/echolab2/processing/integration.py) they basically do integration over pixels in linear domain that fall within the grid computed from grid.py, and then converting it to the log domain
- their grids here are much larger and since it is integration, it corresponds better with NASC, especially because there is "layer" and "interval" in the language, so not really the finer regridding we are discussing here.

lsetiawan, 2023-11-03:
@leewujung Could you point us to a test data for this?

leewujung, 2023-11-03:
I haven't generated any datasets for this. I think to generate outputs from the IMR functions and test against it would be the way to go. The input would be Sv with potentially irregular spacing in ping time and depth across channels, and output would be Sv gridded to the same ping time and depth grid across all channels.

anantmittal, 2023-11-22:
Explore `mock_Sv_dataset_irregular` and `mock_Sv_dataset_regular` fixture in `echopype/tests/commongrid/conftest.py`

anantmittal, 2023-11-22:
@leewujung How should we decide the correct ping time?

anantmittal, 2023-11-22:
> @leewujung How should we decide the spacing in ping time for regridding?

Assuming the `regrid` function takes `ds_Sv`, `range_bin`, and `ping_time_bin` as arguments.

leewujung, 2023-11-24:
@anantmittal : Not sure if I understand the question, but I think in terms of uses, it may be the most useful if users specify `ds_Sv`, `range_wanted`, `ping_time_wanted` as input arguments, and get `ds_Sv_out` with `range_wanted` and/or `ping_time_wanted` as the coordinates across all channels.

One common use case is when different channels have different sizes along the `range_sample` dimension, and a user wants to align them all according to one of the channels. In that case, the call would look something like:
```python
ds_Sv_out = ep.consolidate.regrid(
   ds_Sv = ds_Sv,
   range_wanted = ds_Sv["range"].isel(channel=0, ping_time=0)  # use range of the first ping in first channel
)
```

A couple notes:
- The range values for all pings are very often the same but not always, since for EK echosounders it is something a user can change "on the fly."
- In the the above case `ping_time_wanted` is not specified. It is pretty common to have all channels to ping on the same time base, but more recently the instruments are much more flexible so this may not be true. In that case, users may want to specify that also to get `ds_Sv_out` to align on both `range_wanted` and `ping_time_wanted`.

anantmittal, 2023-12-07:
> In the the above case ping_time_wanted is not specified. It is pretty common to have all channels to ping on the same time base, but more recently the instruments are much more flexible so this may [...]

@leewujung: Could you explain this comment a bit more? It's unclear how regridding/interpolation should work when both `range_wanted` _and_ `ping_time_wanted` are supplied to the `regrid` function. 

#1241 is my stab at implementation when `ping_time_wanted` is `None.`

leewujung, 2023-12-16:
@anantmittal @lsetiawan : This following paragraph may be useful to understand the process? It also points to the specific approach they use and references in iris from this Sec. 2.2.2 in [this paper](https://www.sciencedirect.com/science/article/pii/S0165783622001886?via%3Dihub).

[image]

lsetiawan, 2024-01-12:
xESMF Regrid Methods explanations: http://earthsystemmodeling.org/regrid/#regridding-methods

Actual algorithms in the xESMF Python library: https://xesmf.readthedocs.io/en/latest/notebooks/Compare_algorithms.html

Tutorial on xarray regridding with xESMF: https://xesmf.readthedocs.io/en/latest/notebooks/Dataset.html

leewujung, 2025-10-02:
Now we have the new option to use the [xCDAT](https://github.com/xCDAT/xcdat) package that is built on xarray to regrid!