echostack-org/echopype #732: Unable to open converted file on Windows

imranmaj, 2022-06-18:
On Windows, `open_converted` fails with a `KeyError` on `Sonar\\Beam_group1`. I believe this is because of the way this function in datatree is currently written: https://github.com/xarray-contrib/datatree/blob/b7cdaa0d702b559a2ece763c4f4dfa707401d5a8/datatree/io.py#L14-L19

The native path separator on Windows is "\", not "/", and the function uses pathlib, which uses the native path separator. This causes nested group paths to use "\" on Windows, which is not a valid netCDF4 group path, causing the netCDF4 engine to error out.

imranmaj, 2022-06-18:
This issue is solved by upgrading xarray-datatree to 0.0.6 but that breaks a lot of other things

emiliom, 2022-06-19:
From https://github.com/OSOceanAcoustics/echopype/pull/731#issuecomment-1159617984:

> I was able to run the tests by manually editing the datatree library source code on my machine

@imranmaj when you get a chance could you share those edits here, if they're small? 

I think this issue is an important one. I've flagged it as Milestone 0.6.1.

leewujung, 2022-06-19:
@emiliom : agreed! This is an urgent issue since Windows users would run into trouble if they upgrade to v0.6.0.

imranmaj, 2022-06-19:
> @imranmaj when you get a chance could you share those edits here, if they're small?

With datatree v0.0.4,

**In `datatree\io.py`:**

Change line 17 from

```
    parrent = pathlib.Path(parrent)
```

to

```
    parrent = pathlib.PurePosixPath(parrent)
```

---

**In `datatree\io.py`:**

Change line 25 from

```
    parrent = pathlib.Path(parrent)
```

to

```
    parrent = pathlib.PurePosixPath(parrent)
````

emiliom, 2022-06-19:
Thanks! Wow, that's simple.

leewujung, 2022-06-20:
Work needed, from @imranmaj :
> Besides the `ChildResolverError` -> `KeyError` it looks like most if not all of the work will be in the repr

closed 2022-07-02 by leewujung