echostack-org/echopype #737: Review/change handling of negative `echo_range` in calibration outputs

leewujung, 2022-06-21:
Right now we set all negative `echo_range` to 0 at the end of `compute_range`, which causes problems when trying to combine datasets (#578)
https://github.com/OSOceanAcoustics/echopype/blob/12693a72badcf54a18408e694b72e7b0e6d48023/echopype/echodata/echodata.py#L473

This issue is a reminder to review this and change/fix related operations in calibration.

leewujung, 2023-03-11:
See https://github.com/OSOceanAcoustics/echopype/pull/968#issuecomment-1464282197 for the divide by zero warnings on ek80_bb_complex and ek80_cw_power range tests.

emiliom, 2023-03-15:
@leewujung I've isolated the source of these log10 warnings. They're all in `calibrate_ek.py`.

- For power data, `tvg_mod_range` can be <= 0 in https://github.com/OSOceanAcoustics/echopype/blob/90e4377e7cb46953bf15d8b57f05ad0ab968d4de/echopype/calibrate/calibrate_ek.py#L68
- For complex data,  `tvg_mod_range` can be <= 0 in https://github.com/OSOceanAcoustics/echopype/blob/90e4377e7cb46953bf15d8b57f05ad0ab968d4de/echopype/calibrate/calibrate_ek.py#L378 and `prx` can be <= 0 in https://github.com/OSOceanAcoustics/echopype/blob/90e4377e7cb46953bf15d8b57f05ad0ab968d4de/echopype/calibrate/calibrate_ek.py#L406

Strictly speaking, something like this for both `tvg_mod_range` and `prx` would fix the warning:
```python
x = x.where(x > 0, eps)
```
where `eps` is a very small positive value (eg, 0.00001). But, I don't know what impact that would have on the calculations. On the other hand, those invalid `log10` applications must be leading to `nan` or `-inf` values right now ...

What do you suggest?

emiliom, 2023-03-16:
Can we close this now? I'm not 100% sure that the exact scope of the issue was addressed by #986