echostack-org/echopype #739: Intermittent bucket error during test runs

leewujung, 2022-06-22:
There has been pretty severe intermittent error related to bucket creation (? I haven't looked very closely) in the latest PRs. From a glance it is of a different cause from those in already documented issues.

One example:
https://github.com/OSOceanAcoustics/echopype/runs/6997551890?check_suite_focus=true#step:16:1703

lsetiawan, 2022-07-08:
PR #761 fixes this issue. Have been pushed to both `dev` and `main` as hotfixes.

closed 2022-07-08 by lsetiawan

leewujung, 2022-07-08:
Awesome!! Thanks @lsetiawan! Would appreciate a run-down on what the problems and fixes are!

emiliom, 2022-07-08:
Thanks for the hard work on this! The first problem sounds like it was a significant burden, so hopefully we'll see noticeable benefits.