echostack-org/echopype #740: Enhance update_platform to support more use cases and produce more consistent results

emiliom, 2022-06-25:
I'll add more details later, including an initial PR. In the meantime, I illustrate and document some of the issues in [this notebook](https://gist.github.com/emiliom/9cddce2417713208f40800dc5ba0abb1)

emiliom, 2022-06-26:
See PR #741

emiliom, 2022-06-29:
> I think it will be clearer to use just the first `ping_time` for this. I am thinking of the case when we combine multiple files, say 10, seeing 10 identical lat/lon with the time corresponding to the first `ping_time` of each file seems more intuitive.

I hadn't thought of that case, and it's a good point. Right now this decision (whether to use a single point with the first ping_time or two points) is external to `update_platform`, but I'm thinking that we could build in -- now or down the road -- a special handling of the fixed-location case such that the user could pass a single lat,lon coordinate; then `update_platform` would implement the assumption (eg, single point timestamped to the first ping_time).

> We can perhaps add in the `comment` that the lat/lon are added through update_platform and use the first ping_time as the timestamp.

In this issue I've included a suggestion to add the `history` attribute to every variable that's updated (inserted/rewritten) by `update_platform`. That would take care of the first part of your suggestion. The second part, the use of the first ping_time as the timestamp, could be added there too (for simplicity); or as a `comment` attribute.

Overall, I like your suggestions and reasoning.

emiliom, 2022-06-29:
For version 0.6.1 (short time frame), I think we can focus largely on the fixed-location use case and the issues and clean ups I've already addressed in PR #741. We can leave the other comments I've raised here to 0.6.2.

leewujung, 2023-03-09:
Just wanted to note that this is related to #769.

leewujung, 2023-05-17:
Now #1009 supersede #769.

emiliom, 2023-08-03:
I've gone over this issue, related PR's, and the outcome of `update_platform` in an example. I think everything here has been addressed. The only exception is the issue now tracked in #1009; we'll discuss and track its progress there and can close this issue.

emiliom, 2023-08-16:
Now that I've opened #1126,  we close this issue.

closed 2023-08-16 by emiliom