echostack-org/echopype #742: Coordinate `filenames` of data variable `source_filenames` should be a coordinate

leewujung, 2022-06-28:
I am just working with the calibrated Sv dataset and trying to combine many of them, and I noticed that the `filenames` is a dimension but not a coordinate. This makes it not possible to automatically combining the `source_filenames` together along the `filenames` dimension.

It seems that this data variable is added here and we can make `filenames` a coordinate explicitly to make it easier to work with.
https://github.com/OSOceanAcoustics/echopype/blob/12693a72badcf54a18408e694b72e7b0e6d48023/echopype/calibrate/api.py#L95

@emiliom : tagging you since this is related to what we want to include in the data provenance. This seems like a one-line fix?

emiliom, 2022-06-28:
`source_files_vars` is a function that's reused in at least a couple of places, including here and in `set_provenance`. A general solution would be to change `source_files_vars` so that it returns dicts for both the variable and the coordinate, just like the [_beam_groups_vars](https://github.com/OSOceanAcoustics/echopype/blob/a3ddec1d9f9aa6bbd394aac82c0bdffcfde90477/echopype/convert/set_groups_base.py#L190) method. See this example of the use of that method:
https://github.com/OSOceanAcoustics/echopype/blob/a3ddec1d9f9aa6bbd394aac82c0bdffcfde90477/echopype/convert/set_groups_ek60.py#L198-L199

So, I'd change the line you mentioned into roughly this:
```python
source_filenames_var, filenames_coord = source_files_vars(source_file)
cal_ds = cal_ds.assign(**source_filenames_var).assign_coords(**filenames_coord)
```

Of course, we'll need to modify `source_files_vars`, too.

emiliom, 2022-09-20:
#803 already addressed this issue. I'm closing it.

closed 2022-09-20 by emiliom