echostack-org/echopype #745: Add test for calibration output format consistency

leewujung, 2022-06-30:
From @emiliom 's comment in #735:
> The form of the resulting Sv, after conversion and calibration, should be identical regardless of the source raw data type; if it's not, the test responsible for catching such errors is test_calibrate.py, not this one.

We currently only have tests to make sure the calibration routines are run through, and in a few limited cases the calibrated output values. Let's implement another test for checking the format consistency of the calibration outputs.

MohamedNasser8, 2024-02-26:
Hello @leewujung
When i tried to copy the data from container into test_data folder i found the default containrization name is "docker-httpserver-1"
but it was set by default in the setup-service.py with underscore "docker_httpserver_1".

MohamedNasser8, 2024-02-27:
so i have written this so far is there another attrs to consider or some constrain i should i cover

```
attrs = ['Sv',
         'echo_range',
         'frequency_nominal',
         'sound_speed',
         'sound_absorption',
         'sa_correction',
         'gain_correction',
         'equivalent_beam_angle',
         'angle_offset_alongship',
         'angle_offset_athwartship',
         'angle_sensitivity_alongship',
         'angle_sensitivity_athwartship',
         'beamwidth_alongship',
         'beamwidth_athwartship',
         'source_filenames',
         'water_level']

def test_calibrate(ek60_path):
    ed = ep.open_raw(ek60_path / "ncei-wcsd" / "Summer2017-D20170620-T011027.raw", "EK60")
    ds_sv = ep.calibrate.compute_Sv(ed)

    assert ds_sv.dims.get('channel') == 3
    assert ds_sv.get('Sv').attrs.get('units') == 'dB'
    assert ds_sv.get('echo_range').attrs.get('units') == 'm'
    assert ds_sv.get('frequency_nominal').attrs.get('units') == 'Hz'
    assert ds_sv.get('sound_absorption').attrs.get('units') == 'dB/m'
    for key in ds_sv.keys():
        if key != 'source_filenames':
            assert ds_sv.get(key).dtype == 'float64'
    for attr in attrs:
        assert attr in ds_sv.keys()
    assert ds_sv.get('equivalent_beam_angle').attrs.get('units') == 'sr'
    ```