echostack-org/echopype #750: Not showing echogram using ES80 sonar model

yalonzo, 2022-07-01:
Hello, 

I'm trying to show an echogram with Sonar ES80 data. However, first, I can't open the RAW data with this sonar model. Instead, I try to open with EK80 like parameter and I got success. I don't know if this trick is correct.

Now, I'm trying to show the echogram but I don't see anything (see the image)

[image]

Can you help with this please.

Thank you! 😎

emiliom, 2022-07-01:
Thanks for the report. Could you provide the error you got when you used `sonar_model="ES80"` in `open_raw`?

yalonzo, 2022-07-04:
There isn't a problem when I read the data using `open_raw `with `sonar_model="ES80"`. However, when I want to calibrate the data I get this error (see the image)

[image]

So, this is the principal reason to use EK80 instead ES80 when I read raw data, but the echogram is empty using this trick.

What do I need to do to see the echogram?

emiliom, 2022-07-05:
Thanks for that information and screenshot. What echopype version are you using?

I think there may be two different issues going on. The first one is that it looks like `calibrate.compute_Sv` is not handling "ES80". I see that ES80 is not included here among the defined "calibrators", and it's also not handled explicitly in the blocks below that line.

https://github.com/OSOceanAcoustics/echopype/blob/a20ea8207484ed05c68e6556293c28b447fa0019/echopype/calibrate/api.py#L10

That's probably a relatively easy fix on our end.

The reason the `Sv` plot looks empty is a different problem. Do you have a small raw file you could share with us to test this? Preferably less than 50 MB or so.

yalonzo, 2022-07-05:
I understand. Ok.

I share with you the RAW file that I'm trying to plot.
[sample_raw.zip](https://github.com/OSOceanAcoustics/echopype/files/9047879/sample_raw.zip)

yalonzo, 2022-07-05:
So, can you plot the echogram with this RAW data? How can I plot the echogram now?

Also, I see that will be a new version of Echopype with new fixes, when will it be released?

Will I be able to plot the echogram with this new version?

Thanks in advance 😎

leewujung, 2022-07-07:
@yalonzo : I am pretty sure what you have seen is something caused by the bug that https://github.com/OSOceanAcoustics/echopype/pull/755 addresses. As @emiliom said, we will have to add ES80 to the list so that it can be handled by the same calibration function as the EK80 echosounder.

In terms of timeline, our goal is to release v0.6.1 by the end of this week. Working toward that!

yalonzo, 2022-07-08:
Ohh, I understand. Thank you!!! 😎

leewujung, 2022-07-08:
It is now on conda-forge!

yalonzo, 2022-07-09:
I tried it and ran perfectly. Thank you so much for the support and again Gracias!!! 😎

closed 2022-07-09 by emiliom

yalonzo, 2022-07-27:
Hi again,

One question, can you plot the echogram using the MVBS_ds? I used this function but my kernel in Python died :(.

```
MVBS_ds = ep.preprocess.compute_MVBS(
    Sv_ds,
    range_meter_bin=5,  # in meters
    ping_time_bin='20s'  # in seconds
)
```

I want to plot this, but is necessary the conversion MVBS_ds y_axis (depth), x_axis (time) (see image) 

[image]

emiliom, 2022-07-28:
I don't think I understand exactly what you're asking. Yes, one can plot the MVBS echogram, as you can see from the example and screenshots you included, which I can see are from the [first notebook in our Echopype Examples site](https://osoceanacoustics.github.io/echopype-examples/echopype_tour.html).

Could you create a new issue for this question? That makes it easier for us to follow up, compared to extending a discussion in an issue that's already closed. Thanks!

yalonzo, 2022-07-31:
So sorry, 

I referred that I can't plot the MVBS_ds with the data from ES80 sonar model that I shared with you.

Thank you

yalonzo, 2022-07-31:
Also, please 🙏, could you delete this issue (**Not showing echogram using ES80 sonar model #750**) from GitHub because this information and results are sensible and private.

I hope you understand me :(

Thank you for your help and your comprehension.

emiliom, 2022-08-02:
> I referred that I can't plot the MVBS_ds with the data from ES80 sonar model that I shared with you.

Are you saying you can create `MVBS_ds` using `compute_MVBS` just fine, but you get a kernel death when you try to plot following the Echopype Examples code? If so, please share the exact statement where the kernel death occurs, and preceding statements. Also, how big is your raw file?

> Also, please pray, could you delete this issue (Not showing echogram using ES80 sonar model https://github.com/OSOceanAcoustics/echopype/issues/750) from GitHub because this information and results are sensible and private.
>
> I hope you understand me :(
>
> Thank you for your help and your comprehension.

We appreciate when users like you take the time to post a question here on GitHub and we try to help whenever possible. That feedback makes echopype better, and that benefits the community of echopype users. But because this is an open source software package and our time is limited, our decisions on when and how much to help are based on the common understanding that exchanges on GitHub are public and will remain public, for others to benefit from and for our own future reference.

If you had told us in advance that all our communication would have to be private, we probably would not have offered to help. So I'm afraid we can't offer to delete an issue. However, you can edit specific previous comments you made if you feel there was sensitive information present. I think you've already done that, because I don't see the comment where you shared the sample raw file.

BTW, in my opinion, there is nothing in this exchange that is highly specific to your project or that provides any context about the data or your project. The most specific thing is the echogram plot I made (https://github.com/OSOceanAcoustics/echopype/issues/750#issuecomment-1179165290), but even that has no reference at all about where the data come from or even the date. It's basically a small, generic echogram. You've already removed -- appropriately -- the link to your raw file.

yalonzo, 2022-08-02:
> Are you saying you can create MVBS_ds using compute_MVBS just fine, but you get a kernel death when you try to plot following the Echopype Examples code? If so, please share the exact statement where the kernel death occurs, and preceding statements. Also, how big is your raw file?

Yes, when I tried to get and then plot the MVBS data from the example shared (sample_raw.raw), my kernel in Python died. I don't know why is the reason 🤔.

This is the code used:
```
MVBS_ds = ep.preprocess.compute_MVBS(
    ds_Sv, 
    range_meter_bin=0.2,  # 0.2 meters
    ping_time_bin='10S'   # 10 seconds
)

MVBS_ds
```

And this is the message: 
[image]

> BTW, in my opinion, there is nothing in this exchange that is highly specific to your project or that provides any context about the data or your project. The most specific thing is the echogram plot I made (https://github.com/OSOceanAcoustics/echopype/issues/750#issuecomment-1179165290)

Yes, I understand your point of view, in that case, only could you edit or remove the part of the echogram that you drew?

Thank you so much 😎

emiliom, 2022-08-10:
I've removed the plot I had created -- though, again, there was nothing there that made it identifiable or distinguishable from any echogram from an unknown source.

I'll try to reproduce your kernel error on `ep.preprocess.compute_MVBS`, but it won't be for at least two weeks. I suggest you profile memory consumption when you run that statement to see why and how the kernel dies. Typically a Jupyter kernel death is due to memory usage.