echostack-org/echopype #763: Interpolate locations data to echograms

yalonzo, 2022-07-12:
Hello guys,

Is possible to get the longitude and latitude data from Platform Group (see the screenshot) to the Sv data?
[image]

I see that the latitude and longitude data are correlated to time1. However, the Sv data is correlated to time2
[image]

The latitude data has a similar shape that time1
[image]

And the Sv data has a similar shape that time2
[image]

So, is possible to interpolate this time1 to time2? 🤔

This is because with that will be possible to scan where there are biomass or another species in the echogram according to locations

Thank you 😎

emiliom, 2022-07-12:
We've started working on that feature. See #749. @leewujung can comment on whether there's already useful functionality in that PR.

leewujung, 2022-07-13:
@yalonzo : the function in #749 is already in a functional form, but I want to move it to a different subpackage/module as part of our work in constructing the entire data processing pipeline, so it will remain a PR until probably sometime next week (and later be included in v0.6.2). Feel free to grab that function to use and comment on that PR for more discussion!

leewujung, 2022-07-13:
Duplicate of #725

emiliom, 2022-08-04:
I'm closing this issue, since we already identified it as a duplicate. See #725

closed 2022-08-04 by emiliom