echostack-org/echopype #770: Read CW EK80 error when using transceiver_software_version=50112

rolandproud, 2022-08-05:
Hi, error when reading EK80 (power/CW) from transceiver_software_version=50112. There are no filters/decimation factors so SetGroupsEK80() throws error. Fix by making this optional. Also, Sv is not caluclated correclty using these files. Sound absorption/sound speed only calculated for first ping (due to enviroment values) and the rest are nans and hence calculated Sv is nan. I had to provide:

env_params={'temperature' : ek['Environment']["temperature"].data,
                    'salinity'    : ek['Environment']["salinity"].data,
                    'pressure'    : ek['Environment']["depth"].data}

to make it work. I can provide raw datafile if needed.

Cheers

leewujung, 2022-08-05:
@rolandproud : which echopype version are you using? I think the fixes to these were included in v0.6.1.

rolandproud, 2022-08-05:
Ah sorry - 0.6.0. OK thanks!

closed 2022-08-05 by rolandproud

leewujung, 2022-08-05:
No problem. Let us know if these persist even with v0.6.1!