echostack-org/echopype #790: Add option in `consolidate.add_depth` to accept DataArray or use variables already in EchoData

leewujung, 2022-08-19:
Right now `consolidate.add_depth` accepts only scalar inputs for input arguments `depth_offset` and `tilt`. However, these two variables can conveniently be DataArray with a time dimension. Or we may want to use the variables already stored in the EchoData object from which the Sv is derived from. 

Relevant variables in the EchoData object are `water_depth`, `vertical_offset`, and `tilt` (AZFP only) all in the platform group.

## Tasks
- [ ] allow accepting DataArray as inputs
- [ ] allow using variables already stored in the corresponding EchoData object
- [ ] add tests for the above cases

emiliom, 2023-06-27:
@leewujung: AZFP data come with two tilt variables, `tilt_x` and `tilt_y`. I haven't been able to find unambiguous definitions of these two variables in ASL / AZFP documentation. What are the orientations of x and y?

`consolidate.add_depth` has a `tilt` argument. For AZFP data, how do we calculate `tilt` from `tilt_x` and `tilt_y`? Would the calculation always be unambiguous?

emiliom, 2023-06-30:
I'm in touch with Steve and Julek (ASL) and have made progress with understanding `tilt_x` and `tilt_y`.

emiliom, 2023-08-17:
Refer to #1051 for related discussions.

closed 2024-07-12 by ctuguinay