echostack-org/echopype #796: "WindowsPath" object is not iterable

IanTBlack, 2022-08-25:
I am trying to run through the [OOI EK60 Mooring Example](https://github.com/OSOceanAcoustics/echopype-examples/blob/main/notebooks/ms_OOI_EK60_mooringtimeseries.ipynb) and have received an error `"WindowsPath" object is not iterable` when running the line `ds_Sv = ep.calibrate.compute_Sv(ed)` in the "Convert Multiple Files" section.

I receive this error when running the example as a notebook in Pycharm and when running it line by line in my system console.

My work around for this is to loop through each file individually and append them to a list. ep.open_raw > ep.calibrate.compute_Sv > ep.preprocess.compute_MVBS. Once each file has been process individually, I use xarray.merge to merge all of the datasets together. 
I've had mixed success with this method.

Is there something I can change to avoid the `WindowsPath` error? Thanks!

leewujung, 2022-09-08:
Hi @IanTBlack : Could you let us know what echopype version you are using, whether you install echopype using conda or pip, and provide the full trace back error messages? These will help us with isolating the potential problem. Thanks!

IanTBlack, 2022-09-10:
Echopype Version: 0.6.2
Install Method: pip

Full Traceback:
```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
Input In [24], in <cell line: 1>()
----> 1 ds_Sv = ep.calibrate.compute_Sv(ed)

File ~\AppData\Roaming\Python\Python39\site-packages\echopype\calibrate\api.py:201, in compute_Sv(echodata, **kwargs)
    115 def compute_Sv(echodata: EchoData, **kwargs) -> xr.Dataset:
    116     """
    117     Compute volume backscattering strength (Sv) from raw data.
    118 
   (...)
    199     it must be set using `EchoData.update_platform()`.
    200     """
--> 201     return _compute_cal(cal_type="Sv", echodata=echodata, **kwargs)

File ~\AppData\Roaming\Python\Python39\site-packages\echopype\calibrate\api.py:103, in _compute_cal(cal_type, echodata, env_params, cal_params, waveform_mode, encode_mode)
    100 else:
    101     source_file = "SOURCE FILE NOT IDENTIFIED"
--> 103 cal_ds["source_filenames"] = source_files_vars(source_file)["source_filenames"]
    104 prov_dict = echopype_prov_attrs(process_type="processing")
    105 prov_dict["processing_function"] = f"calibrate.compute_{cal_type}"

File ~\AppData\Roaming\Python\Python39\site-packages\echopype\utils\prov.py:53, in source_files_vars(source_paths)
     51     source_files = [str(source_paths)]
     52 else:
---> 53     source_files = [str(p) for p in source_paths]
     55 source_files_var = {
     56     "source_filenames": (
     57         "filenames",
   (...)
     60     ),
     61 }
     63 return source_files_var

TypeError: 'WindowsPath' object is not iterable
```

emiliom, 2022-09-10:
Thanks for that information and error log. I think I've found the source of the problem. I believe we'll need to handle `WindowsPath` in this line:

https://github.com/OSOceanAcoustics/echopype/blob/6276f6c693f9c5a05e3e162d0ef1648e17d447e3/echopype/utils/prov.py#L49

You're running on Windows, right? Are you able to run the the two other notebooks in https://osoceanacoustics.github.io/echopype-examples/?

emiliom, 2022-10-05:
This bug has been fixed in the `dev` branch; see PR #829. It'll be part of the next echopype release.

@IanTBlack thanks again for reporting this!

closed 2022-10-05 by emiliom