echostack-org/echopype #801: netcdf problem with echopype installed using pip install

emiliom, 2022-09-05:
When using echopype installed using `pip install echopype`, there is a problem when saving an xarray dataset to netcdf. The problem involves the compression of unicode strings, and it's extraneous to echopype proper. It was reported in #789.

I'm describing the problem in more detail here, copying directly from that other issue. Both environments were created with echopype 0.6.2.

This simple freestanding code illustrates the problem:

```python
import numpy as np
import xarray as xr

x = np.array([1,2],dtype='U') #This results in RuntimeError.
#x = np.array([1,2],dtype='f') #This is working well.
da = xr.DataArray(x,name="x")

encoding={'x': {'zlib': True, 'complevel': 4}}
da.to_netcdf(path="output.nc" encoding=encoding)
```

This environment generates the error using the sample code:

```bash
conda create -n echopype_pip -c conda-forge python=3.9 ipykernel
conda activate echopype_pip
pip install echopype
```

In contrast, with this similar environment where echopype is installed from the conda package, the same code runs without errors:

```bash
conda create -n echopype_conda -c conda-forge python=3.9 ipykernel echopype
conda activate echopype_conda
```

erikatbeof, 2022-09-21:
@emiliom Is anyone looking into this?

emiliom, 2022-09-23:
Not yet. Have you tried using conda?

erikatbeof, 2022-10-10:
I went through the process installing conda. Fail to run the demo, echopype_tour.ipynb, though: 

---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
Input In [1], in <cell line: 5>()
      2 from pathlib import Path
      4 import fsspec
----> 5 import geopandas as gpd
      6 import xarray as xr
      8 import matplotlib.pyplot as plt

ModuleNotFoundError: No module named 'geopandas'

leewujung, 2022-10-11:
@erikatbeof : not sure what you're running into there. Try `conda list` to see if you actually have `geopandas` in your environment since the error message is not finding the module.

erikatbeof, 2022-10-11:
I tried that, -of course, and it's not installed. But I got the impression that using conda would be a easy, a 'one' line solution. It appears not to be. In the contrast codas is clunky. I advise you to make your code working well with pip -an abandon conda.

leewujung, 2022-10-11:
Heh, I don't think abandoning conda is the right solution, and if you are reluctant to fix your environment it is your choice though unfortunate.

The problem you're running into is external to the echopype codebase. It is between xarray and netcdf4. To see that, create the following environment that has nothing to do with echopype:

```shell
$ conda create -n min_test -c conda-forge python=3.9 ipykernel -y
$ conda activate min_test
$ pip install netcdf4 xarray
```

Once you have this, do the following like in the first comment:
```
x = np.array([1,2],dtype='U')  #This results in RuntimeError.
#x = np.array([1,2],dtype='f')  #This is working well.
da = xr.DataArray(x,name="x")

encoding={'x': {'zlib': True, 'complevel': 4}}
da.to_netcdf(path="output.nc", encoding=encoding)  # this will error out
```

emiliom, 2022-10-11:
Thank you @lsetiawan ! Your PR #843 (pinning netcdf4 to <1.6) fixes the issue. This fix will make it into the next echopype release, so that `pip install echopype` should work out of the box because it'll install a version of netcdf4 that doesn't introduce this problem.

lsetiawan, 2022-10-11:
Currently now, when I install echopype `dev` branch, the problem doesn't show up:

[image]

leewujung, 2022-10-15:
This is resolved now. Closing.

closed 2022-10-15 by leewujung