echostack-org/echopype #809: Update version conversion from v0.5.x to v0.6.3

b-reyes, 2022-09-16:
In some PRs/issues we have made or will make some small changes to the `EchoData` structure e.g. #677, #806. Since we made these changes, the mapping of v0.5.x to v0.6.3 will be outdated and changes will need to occur in [ep_version_mapper.py](https://github.com/OSOceanAcoustics/echopype/blob/dev/echopype/echodata/sensor_ep_version_mapping/ep_version_mapper.py). I suggest that we add a new routine, say `convert_v060_to_v063` that maps v0.6.0 to the v0.6.3 structure. We could then do the following in `ep_version_mapper.py`:

```
  if (0, 5, 0) <= echodata_obj.version_info < (0, 6, 0):
      convert_v05x_to_v06x(echodata_obj)
      convert_v060_to_v063(echodata_obj)
  elif (0, 6, 0) <= echodata_obj.version_info < (0, 6, 2):
      convert_v060_to_v063(echodata_obj)
  elif (0, 6, 3) <= echodata_obj.version_info < (0, 7, 0):
      pass
  else:
     Do not implemented error
```

Maybe we should then change the name `convert_v05x_to_v06x` to `convert_v05x_to_v060`.

Additionally, tests should be completed for this new structure.

Pinging @emiliom since he is the one working on the mentioned updates.

leewujung, 2022-09-22:
We discussed this briefly in the meeting today and there is a broader and a narrower interpretation of this:
- narrow: there are changes that involve adding new variables, either data variables or coordinates, such as those arise from the `source_filenames` overhaul and the addition of `meta_source_filenames`
- broader: we are likely going to continue adding/changing some variables, especially along the provenance and attributes side of things

For the narrower one, these can be addressed "locally" as checks for existence of variables in specific functions that require them. These can serve as temporary "patch" to make sure new (minor) releases go in without problem.

For the broader interpretation though, we should consider what might a more efficient strategy be -- is doing a version converter for every major release a better way to handle this?

leewujung, 2023-10-25:
I'll close this now since it's not clear whether there is practical value of a data version converter for earlier versions. This is something useful to have once the data format is more fully stabilized.

closed 2023-10-25 by leewujung