echostack-org/echopype #816: A `CombinedEchoData(EchoData)` class?

leewujung, 2022-09-17:
Previously in https://github.com/OSOceanAcoustics/echopype/pull/797#issuecomment-1229790414 I had some thoughts about creating an `CombinedEchoData(EchoData)` object that inherited from the `EchoData` object. I think this can potentially help manage the confusions/problems we have with the `source_filename` attributes. 

@emiliom @lsetiawan @b-reyes : Interested in your thoughts on this. My understanding is that @b-reyes has not integrated the combine-via-writing-to-zarr functions into other parts of the package, so this is the right time to discuss and decide on this.

On top of my head, potential advantages of doing this are:
- a clean separation of `EchoData` that comes directly from raw files, and `CombinedEchoData` coming from multiple `EchoData` or converted files
- the combine-via-writing-to-zarr functions @b-reyes is implementing are cleanly separated from the base `EchoData` functions
- `source_filenames` as it stands now will be length=1 if coming from a single file and length>1 if coming from multiple files
- make it easier for us to deal with the details in the after-combine Provenance group, since we will not be touching the base `set_group`s and `EchoData` code

b-reyes, 2022-09-19:
> the combine-via-writing-to-zarr functions @b-reyes is implementing are cleanly separated from the base EchoData functions

From my current understanding, my implementation will eventually become the only approach used after we depreciate the old combine method. If this is the case, I am not quite sure it makes sense to inherit the `EchoData` class and have `CombinedEchoData(EchoData)`. The way my approach works is that we write the combined files to a zarr and then open the zarr store using `ep.open_converted` and return an `EchoData` object. So I am not quite sure what we would gain by inheriting the `EchoData` class. @leewujung Maybe I am missing something here?

leewujung, 2022-09-19:
@b-reyes : I definitely don't think we need to keep the old combine functions at all! what I meant is that the functions you are working on that are specifically to the `combine_echodata` function, which performs the combinations and also specifically dealing with the attributes (in the next release) can be cleanly separated by returning an inherited class. This way we have clean separation of what we _need_ to have for the combined outputs, and what we already have in the existing `EchoData` class.

I think that `ep.open_converted` can return:
- either an `EchoData` object if it is from a file/store converted (but not combined) from raw files,
- or a `CombinedEchoData` object if from a file/store that are results of `combine_echodata`

This way our handling of provenance would be more trackable.

b-reyes, 2022-09-19:
> I think that ep.open_converted can return:
> -  either an EchoData object if it is from a file/store converted (but not combined) from raw files,
> - or a CombinedEchoData object if from a file/store that are results of combine_echodata
> This way our handling of provenance would be more trackable.

Ok, now I think I see what you are saying. I will have to think about this a little more, but I do see there are some advantages here.

leewujung, 2022-09-19:
Sounds good. There must be some ramifications and I will think on this some more too.

emiliom, 2022-09-20:
That's an interesting suggestion. The advantages you list boil down to a certain tidiness to where the combine functionality would be found (as elements of the `CombinedEchoData` class), and a potentially cleaner and more reusable (or at least "inspectionable", to make up a word)  way of defining and handling the special, extra Provenance information (relative to SONAR-netCDF4, as found in the current `EchoData` class). Right now all the combine functionality is a set of functions in the `combine.py` module, and the default would be for @b-reyes' new functionality to be structured that way, too.

Potential downsides that come to mind are:
- One more class to wrap one's head around (though it'll be largely `EchoData`)
- `open_converted` will need to be able to determine whether it returns an `EchoData` or `CombinedEchoData` object. Instead of simply opening all groups, as currently done, it may need to inspect something in the Provenance group (a particular variable or attribute) to decide what to return.
- Wherever there are tests for whether an object is an `EchoData` object, now those tests will also need to check if the object is a `CombinedEchoData` object, or more generically, whether its `type` is a subclass of `EchoData`.

There is a clean logic to defining a combined dataset as a specialized type of `EchoData` object, primarily with respect to Provenance. What is actually different about a `CombinedEchoData` object, other than the functionality for creating the object and the extra Provenance entities (variables and attributes), may be tiny. Maybe for downstream use, the existence of a distinct subclass mainly serves as a clean shorthand for making processing decisions regarding Provenance: If it's an `EchoData` object, do this; if it's a `CombinedEchoData` object, do that.

Overall, I think I see modest gains with modest costs. I support the idea, unless we identify bigger costs.

Finally, regarding this point:

> - `source_filenames` as it stands now will be length=1 if coming from a single file and length>1 if coming from multiple files

I think that's neither here nor there. `source_filenames` will remain unchanged regardless. The convention itself already supports the current usage, with length >=1.

leewujung, 2022-09-22:
We discussed this and decided that we will NOT create a separate `CombinedEchoData` class but will do the following:
- add a attribute in the `Provenance` group to signify that the dataset is resulted from combining multiple EchoData object
- potentially implement additional functions to sort through the `Provenance` attributes in later releases

leewujung, 2022-09-22:
I'll close this based on the above decisions.

closed 2022-09-22 by leewujung